Klipper3d/klipper #3097: install-octopi.sh needs to check / specify python version for venv creation

experimentech, 2020-07-16:
There is no log file because there isn't one.
Not all devices default to Python2.x venv. I have a system running as an OctoPrint server (Orange Pi Zero, Armbian Buster) which defaults to Python 3 for venv.
install-octopi.sh (and possibly other scripts) blindly install packages and states it has started the klipper service. In reality it has failed silently.

experimentech, 2020-07-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4929659/klippy.log)
Addes a shill klippy.log to get past the gitbot. A log file is not generated because of the initial failure due to install-octopi.h attempting to run Python2.x code in a Python3 venv, because it does not check or set the version of Python before proceeding.

mdealer, 2020-07-18:
Having the same problem here. The Klipper service just fails with `Traceback (most recent call last):
  File "klipper/klippy/klippy.py", line 8, in <module>
    import util, reactor, queuelogger, msgproto, homing
  File "/home/user/klipper/klippy/reactor.py", line 6, in <module>
    import os, select, math, time, Queue as queue
ModuleNotFoundError: No module named 'Queue'
`

KevinOConnor, 2020-07-19:
What octopi image did you use?

Also, please don't spoof the bot - I'll remove the "incomplete issue" tag if the log is not needed.

-Kevin

The-Monkey-King, 2020-07-25:
> 
> 
> Having the same problem here, but with install-ubuntu...sh. The Klipper service just fails with
> `Traceback (most recent call last): File "klipper/klippy/klippy.py", line 8, in <module> import util, reactor, queuelogger, msgproto, homing File "/home/user/klipper/klippy/reactor.py", line 6, in <module> import os, select, math, time, Queue as queue ModuleNotFoundError: No module named 'Queue'`

Something similar when I tried to install klipper on Debian 10.4. I manually installed queue, cffi, serial . Still no hex file, but that could be because the original installation failed. See: [https://github.com/KevinOConnor/klipper/issues/3123](url)

mdealer, 2020-07-31:
@KevinOConnor In my case it's Ubuntu 20.04 LTS, no Raspberries, but a real PC. I suppose it can happen on any Linux with Python 3 as default.

KevinOConnor, 2020-08-05:
Ah, okay.  Well, the install-octopi.sh was only really intended to run on the OctoPi Linux distribution.  I agree the script should be updated though.

-Kevin

inks007, 2020-08-09:
I've noticed the same thing (Ubuntu64 20.04LTS on RasPi4). The workarounds suggested elsewhere are working, but here is the lot:

```
sudo apt-get install -y python2
cd ~
git clone https://github.com/KevinOConnor/klipper
sed 's/python-virtualenv/python3-virtualenv/' -i ~/klipper/scripts/install-ubuntu-18.04.sh
virtualenv ~/klippy-env --python=python2
source ~/klippy-env/bin/activate
~/klipper/scripts/install-ubuntu-18.04.sh
deactivate
```

1. This installs python2 alongside python3.
2. It deals with apt complaining that _Package 'python-virtualenv' has no installation candidate_.
3. It sets up the python2 virtual environment for klippy, which allows the install script to pip install the python dependencies correctly.

@The-Monkey-King it is easiest to remove klippy-env, then try the above (but maybe use install-debian.sh)

eduncan911, 2020-08-28:
@mdealer @The-Monkey-King i was able to walk through the Ubuntu script to get it into 20.04 with Python 3.

Documented it in #3256

KevinOConnor, 2020-09-03:
I added the "-p python2" to the virtualenv command of install-octopi.sh (and similar) in commit 81865b59.

-Kevin

Fabrizio3, 2020-09-19:
i believe its failing right at the start of instalation, ill open a separate issue as not to egt confused, been 3hs searching around in the issues and googling with nothing coming up

closed 2020-10-01 by KevinOConnor