Klipper3d/klipper #3098: Question: Custom control loops

Daniel-Kurth, 2020-07-16:
Hey, I have a question.
Is it possible to implement custom control loops using Klipper?
Ideally the loop would be executed at regular intervals (is there a limit to the update frequency of such a loop) and it would process sensor data (is there a limit on how fast sensor data can be acquired) and based on the sensor data, the output would be adjusted (what kind of delay would there be until the processed data on the RPI reaches the output pin of the mcu?).

Thanks for the help.

ld3300, 2020-07-17:
You can run a periodic loop with [delayed_gcode].  How fast of a frequency are you looking for?

With [delayed_gcode] you can give it an initial value if you want it to start executing when Klipper restarts, or start it with a print command, then build into itself an update frequency, that could also be dependent on if statements, you would call `UPDATE_DELAYED_GCODE [ID=<name>] [DURATION=<seconds>]` either from within the delayed_gcode to keep it executing or you could set it externally. if you set the duration to 0 it cancels the loop until you give it an update time again.

KevinOConnor, 2020-07-19:
It is possible to implement custom control loops.  It's possible to do them in the C micro-controller code (see src/endstop.c as an example).  When implemented in the micro-controller the round-trip-time is generally measurable in micro-seconds.  It's also possible to implement them in the host python code (see klippy/extras/heaters.py as an example).  When implemented in the host code, we generally use a round-trip time of 100ms or greater.  As indicated earlier, it's also possible to implement a control loop in g-code macros using the [delayed_gcode] system.

-Kevin

Daniel-Kurth, 2020-07-20:
Thank you very much!

closed 2020-07-20 by Daniel-Kurth