Klipper3d/klipper #3100: PROBE_CALIBRATE works different to explained in documentation

cyrix2000, 2020-07-16:
Hi,

When I'm trying to calibrate my z offset (using PROBE_CALIBRATE with BLTouch probe) the documentation (https://www.klipper3d.org/Probe_Calibrate.html) states that the nozzle will move to the previous probe position, to then enable manual z adjustments. 

In my case, it is doing the opposite, the probe moves to the previous nozzle position (so exactly the opposite what it should do according to the documentation).

Have I configured something wrong here, or is the documentation not dedcribing the command behaviour (in my opinion it makes sense what the documentation is saying, as you want to do the "paper test" at the posiiton the bed has been probed previously, so moving the nozzle further away from there wouldn't make much sense).

[klippy.log](https://github.com/KevinOConnor/klipper/files/4934049/klippy.log)

jakep82, 2020-07-16:
Is it possible you reversed the nozzle and probe numbers when calibrating your X and Y offsets?  Your config currently indicates the bltouch is 44mm to the left of your nozzle, and 7mm in front of it.

cyrix2000, 2020-07-16:
Yes, that's correct. The probe is mounted to the left and in front of the nozzle (I used the same offset before for Marlin and it worked fine).

KevinOConnor, 2020-07-19:
If the nozzle does not move to the position that used to be directly below the probe then the x_offset and/or y_offset settings are not correct.  You'll need to modify them.  Klipper is not Marlin, so don't try to copy these values unmodified.

-Kevin

cyrix2000, 2020-07-19:
Hi Kevin,

I followed your documentation in getting the x,y offset for the BLTouch step-by-step and entered the values in the config. The probe is sitting to the left and in front of the nozzle with (0,0) being in the rear right corner of the priinter bed (Ender 5). So I have two negative values for x and y offset.

That should be correct?

BlackStump, 2020-07-19:
The general accepted origin (0,0)is front left

KevinOConnor, 2020-07-19:
Yeah - it seems your axes are backwards and that's causing confusion.  Best to fix the axes.  This is likely related to issue #3072 .

-Kevin

cyrix2000, 2020-07-20:
Ok that seems to be the problem. Will change the origin, that will I guess also fix the troubles I have with the bed visualiser in Octoprint :). Thanks for the help.

closed 2020-07-20 by cyrix2000