Klipper3d/klipper #3102: Question: BLTouch needle at BED_MESH_CALIBRATE as for PROBE_ACCURACY

SphaeroX, 2020-07-17:
Hi, can I set up Klipper so that when the BED_MESH_CALIBRATE is called, Klipper queries the points as in PROBE_ACCURACY?

So take the needles off the BlTouch, move them onto the printing bed, then move them up a bit and then back down again to measure.

Background of the question is with PROBE_ACCURACY I get the following values:

sample at 110,000,110,000 is z=1,915000
sample at 110.000,110.000 is z=1.912500
sample at 110.000,110.000 is z=1.912500
sample at 110.000,110.000 is z=1.912500
....

The first measured value is always different, because it is the first time it goes down, so I would like to take only the second one. With samples you can set how often it measures, but I would like to leave out the first value completely.

SphaeroX, 2020-07-18:
didnt need but heres the log
[klippy.log](https://github.com/KevinOConnor/klipper/files/4942174/klippy.log)

KevinOConnor, 2020-07-19:
Most of the PROBE_ACCURACY parameters are also available in the [probe] config section, and are also available on the BED_MESH_CALIBRATE command - see: https://www.klipper3d.org/G-Codes.html and config/example-extras.cfg .

-Kevin

SphaeroX, 2020-07-21:
Thank you for your answer, I think I misspoke.

I meant that Klipepr at BED_MESH_CALIBRATE should go to each point twice, discarding the first measurement. For the reason that the first measurement usually still has a deviation (for whatever reason).

For example, with Safe z Homing, it does it right. The printer goes down once until the BlTouch triggers, then it goes up again by the length of "sample_retract_dist" and then takes the measurement.

something like this would be perfect during leveling!

KevinOConnor, 2020-07-25:
The common way to handle this is to use multiple samples, set a samples_tolerance, and use a samples_tolerance_retries (see config/example-extras.cfg for the details).  That said, you might want to reduce your probing speed so you don't get these types of errors in the first place (and/or investigate why the first probe is not accurate).

-Kevin

SphaeroX, 2020-07-26:
Thanks a lot, sounds like a good solution to the problem. I have adapted the Config accordingly.

Thanks for your work and all those who help!

> samples: 3
> sample_retract_dist: 1.0
> samples_tolerance: 0.006
> samples_result: median
> samples_tolerance_retries: 3

closed 2020-07-26 by SphaeroX