Klipper3d/klipper #3103: Question: Does Klipper support the A4950 stepper drivers/BTT S42B Closed Loop?

zpatten, 2020-07-18:
First off I love klipper; amazing work and props to everyone involved.

I apologize for posting this as an issue however after a ton of googling and digging about the internet I can't seem to find an answer to this.

I'm looking to use the `BTT S42B V1.0 Closed Loop Driver Control Board` with the SKR v1.4 Turbo board on a printer while doing higher speed printing.  Their specs state this uses an A4950 stepper driver which as far as I can tell is a PWM driver.  I did find a mention in the klipper docs that says:
```
There is also support for current control of traditional stepper drivers via AD5206, MCP4451, MCP4728, MCP4018, and PWM pins.
```
However here it only says `PWM pins`.

And the BTT closed loop driver plugs into a driver socket like so and I'm not sure if this qualifies as `PWM pins`:
[image]

So my basic questions are:
1) Can klipper drive steppers using this system?  Would this just fall into the PWM category?
2) If it can will there be any downsides, loss of functionality, general sadness in klipper over using something like the TMC2209s (which I currently use) - outside of the obvious things like the TMC features (i.e. StealthChop, etc.) and ability to control those features via the klipper configuration.

References:
- https://www.digikey.com/en/product-highlight/a/allegro-microsystems/a4950-dmos-full-bridge-motor-driver-ic
- https://www.biqu.equipment/products/bigtreetech-s42a-42-stepper-motor-closed-loop-driver-board-with-oled-display
- https://github.com/bigtreetech/BIGTREETECH-S42A-V1.0

BlackStump, 2020-07-18:
I think there is support in a branch of klipper [here](https://github.com/KevinOConnor/klipper/tree/work-mechaduino-20181205)
tho it has not made into the main klipper branch
[docs](https://github.com/KevinOConnor/klipper/blob/work-mechaduino-20181205/docs/Mechaduino.md)

zpatten, 2020-07-18:
Awesome, thanks for the information!

That looks promising after reviewing things; minus the giant experimental use with caution warnings 😄 

If I'm willing to contribute and help drag this functionality across the line is this something klipper would be interested in merging into the mainline if we can move it to a stable phase?

I only ask because I'd hate to make such a huge time investment into something like this only to have it sit in a branch unmerged indefinitely.

I have about 40 years of programming experience all the way from BASIC to linux kernel modules and embedded systems as well as similar hardware experience to match.  I will admit as far as SBC's go I'm only super familiar with the Pi side of things so I might have some "dumb" questions regarding the arduino side if I somehow get stuck on something or want/need advice.

Currently I have two printers I'd be will to use as 'test subjects' to work on this with and don't have an issue if we end up frying a few stepper motors or BTT closed loop drivers along the way.

All in all I love klipper; it's made my 3d printing life so much easier and is a beautiful concept all in all; so I would be more than willing to make the investment and contribute on my end assuming that contribution doesn't rot in a corner 😄

All that being said if there isn't much desire to mainline this support I'm perfectly happy using klipper as is with my TMCs.

So let me know how you guys feel about this and if you're interested I'll go ahead and order the BTT hardware.

robthide37, 2020-07-18:
how much was the btt hardware id be willing to dive in if not crazy . I had looked at the mks stuff which would connect right to the skr board without needing a stepstick at all

robthide37, 2020-07-19:
i saw your link but i am curious if could use a different stepper with that encoder. I don't think downgrading stepper quality is wise when we could have way better positional accuracy with .9 nema 23 steppers or even my current ldo .9 1.7 amp steppers. even just 60mm .9 would be great but the railcore designer uses 225 oz nema 23s and explains how the accuracy is much higher because at the low torque of 16x micro stepping he still has enough torque to move the carriage.

zpatten, 2020-07-19:
The kits themselves from BTT are around $15 each.  I went ahead and got 5 of them from BTT off ebay however they are coming from China (and I'm in the US) and lately it's taken things 1-3 months to get to me from China.

I'm using 1.8 nema 17 steppers (2.1 amp for the X and Y), E3D Hemera stepper for the extruder (only gonna stick a closed loop on that if it works reliably on my more 'disposable' steppers) and the rest are currently stock so I don't have any 23's to play with.

Regarding the BTT kits, they just go onto the back of an existing stepper so I think you can use them with the nema 17 0.9's; at least I don't remember seeing anything in their docs about it not working with those (don't quote me on that thou).  

It's possible they will work for the 23's.  An obvious issue will be the difference in the mounting holes; you can't really add much distance between these boards and the steppers as the magnet has to sit in-between them and not move.  It's also possible that differences in the magnetics on the larger motors might not play nice with the magnetic encoder chips used on these boards and/or require a different magnet to be used in-between the stepper and closed loop board.  Given all these add-on closed loop boards seem targeted at the nema 17's I suspect it could be problematic to try and adapt them for larger ones however it's probably worth a try.

Also, based on the GitHub issues for the BTT closed loop drivers it's probable they are janky which likely explains why they are so cheap.  That being said I have a lot more confidence that they will work with klipper running on them instead of the BTT supplied firmware.  All-in-all if they are a bust I will grab some of the actual Mechaduino boards and give those a try.

KevinOConnor, 2020-07-19:
Those "servo steppers" devices can run on an unmodified version of Klipper - they are designed to be a "drop in" replacement for a standard stepper driver.  No special configuration is required in Klipper.

The branch mentioned above is an attempt to run the Klipper micro-controller code on the "servo stepper" directly.  It is not "production ready" and it is not necessary to use.

Code contributions are welcome.  Unfortunately, it's not possible to say if a contribution will be merged before seeing said contribution, however.

Cheers,
-Kevin

zpatten, 2020-07-19:
> Those "servo steppers" devices can run on an unmodified version of Klipper - they are designed to be a "drop in" replacement for a standard stepper driver. No special configuration is required in [...]

Hey Kevin, thanks for the clarification it is appreciated.

I was definitely curious if they would work with Klipper as is since they supply those adapter boards to plug into a stepper driver slot; so again thanks for letting me know that they will work as is with Klipper without modification.

Given my experience with Klipper to date I would honestly feel more confident and comfortable knowing that Klipper was driving those boards instead of say BTT's firmware.

Regarding the code contributions; I totally understand what you are saying.  I did not mean to make it sound like I was wondering if you would commit to merging anything before reviewing it; or anything along those lines.  I am the DevOps Architect and a Lead Software Engineer at the company I work for so I totally appreciate where you are coming from in that regard.  To try and clarify my ask; I was more curious if there is interest in general to moving this functionality to a production ready state.  I would never expect anything I wrote to get merged unless it passed review and met the standards and requirements you and the other core Klipper developers have.

All that being said; if there is a way I could help contribute to Klipper I would love the opportunity to do so.  I can not imagine using any other firmware going forward and at the minimum you've gained a Klipper user for life and Klipper evangelist.

KevinOConnor, 2020-07-19:
Thanks.

There's certainly interest in running Klipper directly on these types of "servo stepper" devices.  I'd certainly prefer to run Klipper on them then a 3rd party firmware of unknown origin.  Be aware there's a bunch of complexity in running them well though - for example: the stepper current pid can result in "salmon skin" effects, some boards now use the stm32 chip, not all the new boards have the usb connector, the a4950 "runs hot".

Also, note that @Arksine has also looked at this recently - https://github.com/Arksine/klipper/tree/work-mechaduino-updated .

Cheers,
-Kevin

zpatten, 2020-07-20:
Indeed.  Ya the BTT board uses the stm32 and luckily has the usb port as well.  

Looks like @Arksine (as well as yourself) has done a ton of work on pushing this forward.  I still need to look over the code changes in depth I only skimmed the diff so far.

I certainly don't want to step on any toes; considering the work that has already been done by yourself and @Arksine do you have any preferences/suggestions on how to proceed.  Seems logical at this point that I should be coordinating at the very least with yourself and/or Arksine.

Is there any documentation/notes/design specs on what else needs to be done; that is unresolved issues with the code as it stands, testing which still needs to be done, further implementation which needs to happen, etc. ?

It will likely be one to two months before my BTT boards arrive from China however I can at least use the time until then to get myself up to speed on the code and do research on the next steps implementation wise.

robthide37, 2020-07-21:
The mks supposedly cloned the mechaduino stuff horribly . I would love some
clear paths but this could be real good alt

Arksine, 2020-07-21:
I haven't had  a chance to work with the mechaduino branch in a while, but I did manage to get basic closed loop stepping to work.   IIRC, there were still some outstanding issues that need resolving:
- The initialization procedure still may need some work.  The test platform I used was the Y axis on an Ender 2.  This was intentional as its poorly designed, the load is uneven because the belts are not mounted parallel to the bed.  When initialized under load the stepper can get confused, resulting in it moving in the wrong direction.
-  As Kevin mentioned, the "salmon skin" effect is obvious.  A different PID implementation or a tuned PID might improve this, I did not spend time attempting to tune the loop as my initial goal was simply to get it working.
- A way to share endstops between multiple servo steppers is necessary if you wish to support anything beyond basic cartesian setups.

I would recommend that you double check the type of board you are getting from BTT.  As far as I am aware, BTT has two variants:  The S42A which is a near clone of the nano zero stepper, this version is equipped with USB.  The S42B is BTTs own take on the servo stepper.  This variant does not include USB, however it does seem to break out a UART so its possible you might be able to connect to klipper that way.  The S42B is also equipped with a different magnetic encoder, support for it will need to be added.

zpatten, 2020-07-21:
@Arksine Thanks for the details.

Regarding the BTT boards I purchased, I spoke too soon and as you pointed out I did get the S42B's which use a JTAG/SWD interface.  I wrongly made the assumption that because some of their documentation referenced a serial interface that it was just plain USB.

That being said using something like the JLink you can apparently get a USB serial port to communicate with the device (at least based on the BTT documentation).  Currently I have no idea if Klipper will work with this as is or if it will require additional coding for Klipper to talk to that serial interface.

There's another factor too as with the added cost of needing a JLink for each BTT S42B (assuming it would even work) that pushes the price point up close to the mechaduino boards.

With all that in mind I will go ahead and grab a JLink so I can at least play around with trying to make it work; I've been meaning to pick one up anyways since venturing into the world of arduinos.

At this point I'm going to consider just grabbing some of the mechadunios; seems like they will be the path of least resistance to being able to work on this and as Kevin pointed out the BTT S42B's can still work with Klipper as is (thou I will be sad knowing Klipper isn't powering them directly).

zpatten, 2020-07-22:
Just to update; I've gone ahead and purchased a mechaduino so at the very least I'll be able to continue the work @Arksine has done.

WildSharkG, 2020-07-27:
Since you're using an SKR v1.4 turbo, you shouldn't worry about the stepper driver because that board has specific pins called CLS for the Closed Loop Stepper motor (no need to change the driver), here's the manual:
[image]
My 2 cents

robthide37, 2020-07-27:
he wants to change the firmware on the stm chip on the stepper . any board can really send the step and dir signal just would need to make your own cable for it. they jus were smart enought to market it to the public meanwhile anyone using cls prob could wire up to any ramps board if they wanted very easily

yschroeder, 2020-07-28:
I just stumbled upon this thread when looking through the Klipper issues (I do this for fun, don't judge me :smile:).

The idea to integrate these cheap boards is very promising! I once thought about getting mechaduinos but their availability was mediocre (shipping to Germany) and the price rather steep to equip all axes. Having these closed loop drivers run Klipper makes wiring a lot nicer: USB and power). They could work for sensorless homing directly, there can be benchmarking to test each axis for maximum speed/acceleration...

For the cheap boards without USB: There will likely be a UART on some pinds (did not check). Attaching some FTDI to those and off you go with Klipper.

For using the JLink serial device: If the JLink serial just shows up as a tty device in /dev/tty* Klipper should be happy to use it.

KevinOConnor, 2020-07-28:
Yes, lots of possibilities with the mechaduino (and similar).  FYI, I wrote about some possibilities last year: https://www.patreon.com/posts/klipper-and-23718908

-Kevin

yschroeder, 2020-07-28:
You mentioned a lot of interesting stuff in that post. For me the most interesting features would be very easy cabling and pushing cheap steppers to their limit by reducing the overengineering as described in your post. USB-C power delivery anyone? One single USB cable, no end stops, nothing!

zpatten, 2020-07-29:
Ya hopefully the JLink will be able to give a USB device for the BTT S42B.  I did look at some of the FTDI devices on Amazon however; I went with the JLink because it seems to be what BTT was using to get a serial interface in their documentation (their screenshots show a JLink-based USB device in addition to a USB device for the firmware flashing in device manager on windows).  So from what I could tell I should get a USB device which will let me push firmware to the S42B and another which can be used to talk to the serial interface.

I've never messed with the JTAG/SWIG interfaces so please excuse my ignorance if they are the same as the FTDI; from what I could tell the FTDI seemed to have a different pinout than the JTAG/SWIG.

@WildSharkG Ya, I'm aware you can just run them straight to the SKR boards; however as @robthide37 mentions I would prefer to run Klipper on them and have the klipper daemon control them via USB.  I'd take Klipper firmware on my boards any day over 3rd party firmware.

yschroeder, 2020-07-29:
I just ordered a S42B myself (delivery in one month). First, I am curious what the default firmware does and then I have a test device for your upcoming Klipper port 😄

I noticed that it has RX and TX pins marked, so attaching an FTDI if all else fails should be easy. Also, here is the original firmware:
https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/tree/master/firmware/S42BV1.0

Personally, I don't have time to help developing this, but count me in as a tester! I have an Ender 3 running Klipper and I am happy to try the S42B with Klipper firmware for any of the axes...

robthide37, 2020-08-02:
i am tryong to decide if 1.8 cls will be more accurate than my .9 non cls .I am feeling that if they are correcting to the full 16x microstepping or even 32 which is what is listed as max , then if the encoder is reliable at that resolution they should be at least 4x more accurate than .9s since you really don't have much past quarter stepping and reallly shouldn't count on more than half so if ran 32x could possible be 16x better layer stacking which would be better than railcore on 225oz nema 23s. That setup has no layer lines visible fromj the naked eye.

closed 2020-09-01 by github-actions[bot]

yschroeder, 2020-09-15:
I got my S42B now, but did not find time to tes tit yet. Did anything happen related to porting Klipper? I am still willing to help testing!

jammi, 2021-08-28:
I installed this on my Klipper device and perhaps the step rate is too fast for it.