Klipper3d/klipper #3104: pid_calibrate interrupted dragon hotend

jorgegarcian, 2020-07-18:
hello, i just installed klipper on a mini skr v1.2 and i was trying to make the pid at 245º also probe at 265º from the hotend dragron but it never ends up giving error and i don't know the reason, i put the log and a capture of the temperature.

Thanks
[klippy.log](https://github.com/KevinOConnor/klipper/files/4940970/klippy.log)
[image]

Sineos, 2020-07-18:
https://github.com/KevinOConnor/klipper/blob/1c508c38aa424f914c4c9843330e0f4849e4d26f/config/example-extras.cfg#L821-L840
Try to increase `max_error` and `check_gain_time`

jorgegarcian, 2020-07-19:
Hello I'm putting 
[verify_heater heater_config_name]
max_error: 150
in the file printer.cfg but I get an error in the header I don't know if it's right in that file or I have to uncomment it in klipper / config / example-extras.cfg

jorgegarcian, 2020-07-20:
Thank you very much, I increased the times as you indicated and it worked without problems

[verify_heater extruder]
check_gain_time: 50
max_error: 150

closed 2020-07-20 by jorgegarcian