Klipper3d/klipper #3105: Question: Is it possible to send a command to move z while y or x is moving

leonmelson, 2020-07-18:
Hi I am trying to find out if Klipper could do Torch Height Controlling movement for a cnc plasma table.

The plasma cutter sends up and down commands based on the Arc voltage so I am trying to find out if I could make something to handle it in Klipper.

An example would be I send a command to move Y 200mm while this move is taking place I would like to be able to send another command to move Z up or down.

I have tried FORCE_MOVE and GCODE_OFFSET MOVE=1 but they are only executed after the move is completed.

Would it be possible to create a delayed_gcode macro to monitor 2 pins to check if Z should move up or down and then send a up or down command on a higher level?

If this is not possible I will try to make an arduino based middle man to collect step_pin:, dir_pin: and enable_pin: to drive the motor but also to give me direct control to the motor to override movement if required for the Torch Height Controller.

The Post Processor of the cam software will be setup to do a Z home before each cut is started so if the command is sent to disable the plasma cutter and it move Z up and over in X or Y more then a specified amount it will send a G28Z before it moves down and start the plasma cutter.

Thank You
Leon Melson

Hywelmartin, 2020-07-18:
Look at manual steppers

leonmelson, 2020-07-19:
Thank You

I will give it a try.

KevinOConnor, 2020-07-19:
If you want to move just the XY carriage, typically one would issue a command like `G1 X50 Y50`.  If you want to move the XY carriage while also moving the Z, one would typically issue a command like `G1 X50 Y50 Z10`.

If you don't really have a Z and are instead trying to move another stepper in parallel than, as indicated above, the manual_stepper config section may help.  There is a limited ability to run manual_stepper commands without synchronizing them to the command stream.  However, anything more complex may be simpler to implement in host python code - the host python code can certainly schedule simultaneous movements of any complexity.  See klippy/extras/manual_stepper.py for an example.

-Kevin

leonmelson, 2020-07-20:
Thank You Kevin for the advice

I will look at changing manual_stepper.py sync_print_time function to allow klipper to send move command in parallel with another move. 

If this works I should just be able to push commands to a extra motor also running on the Z axis to control the cutting height correct?. Or is there a way to do a forced Z move to override the currect Z height in parallel with another X or Y move.

There will be a Z axis in the system but the Z will be moved up and down based on the Arc voltage reading so it could be that the system is moving in one direction X or Y with no command to move Z in the gcode. So the micro controller will need to monitor the up and down pins and then move the Z accordingly. Will something like this be possible from a macro or a delayed macro with a Force Move to Z or do you think a extra Z motor will be required?.

Leon Melson

mental405, 2020-07-24:
So this is like... a CNC plasma cutter or additive MIG welder or something?

You could write a module that adjusts the Z offset based on an analog pin input. Using the Z offset would keep it independent of the GCODE Z value and allow it to be varied on the fly. Though it might get tricky adjusting it quickly enough.

KevinOConnor, 2020-07-25:
It sounds like you want a "control loop" to alter the Z motor position.  There was some discussion of control loops recently in #3098 .  Where a control loop is implemented (host code or micro-controller code) will depend on the desired response latency.

FYI, I think @proto3 has looked at something similar.

-Kevin

proto3, 2020-07-26:
Hey, someone invoked me ?
I see you are trying to implement a plasma THC in Klipper. I'm glad to announce I've recently finished that on my fork (https://github.com/proto3/klipper-plasma). You can basically setup a PID like controller on the axis you want through configuration file, then you can enable/disable it with M6/M7 commands. When THC is enabled the stepper will try to stay as close as possible from the target position (voltage here) while staying inside a specified range.
Another thing you may want on a plasma machine, is the ability to react instantly to the arc transfer event. This is normally not possible with Klipper, but I introduce some clock tricks to make this possible and it works fine now.

Here is Klipper controlling my plasma CNC with THC:
https://www.youtube.com/watch?v=CSa9FMtMxFU

FYI I've also created a CAM software called Sheetah that generate Gcode and connect to Klipper. Klipper streams arc voltage, Z position and XY plan speed in real time, so Sheetah can display this in a graph. This is pretty useful when trying to fine tune your THC settings.

Note this is still not a stable release.
I'm currently working on documentation pages, a YouTube channel and some more stuff to give the project visibility. I really hope to make plasma a bit more open source like 3D printing is.

Sorry for the off-topic ^^

leonmelson, 2020-07-27:
Thank You I will have a look

leonmelson, 2020-07-28:
I have created my first Python script/program called pyGCodeLoader to send a file/URL or string to my printer or cnc and now I want to add a plasma cutter.

https://github.com/leonmelson/pyGCodeLoader

The CNC works fine with stock klipper excluding any extruded config

I do not think any changes to klipper will be required for the plasma cutter only some Macro's. I am going to create a THC with 2 lasers that will transmit a signal to a Photo Diode for isolation from the plasma cutter then on the receiving side there will be an arduino that will monitor the signal and then takeover Z movement "So this will only work with external stepper drivers".

In the future this could be controlled by Klipper. 

So the Z pins enable,step and dir will be sent to another arduino that will by default pass the signal on except if it receives an up or down command from the laser THC.

pyGCodeLoader also allows you to specify a wait command and what that wait command should contain = ok or not wait and just send the code but this will run into Klipper's buffer. Before I used php to push the gcode to klipper but it was not fast enough so that is why I tried to make a python script for this and the speed is allot faster so fast I have not run into a limit yet.

I am new to python but I really like it I have been programming with PHP for years and I have created a FrontEnd called KRoBoT it allows you to create multiple Klipper Robots and allows you to send GCode to them with pyGCodeLoader based on the selection. This is not on Github because it has changed allot over time and it has allot of functions that is not used anymore so need to be cleaned. Two of my friends and I have been using it as our primary system for a while and I will try to make it cleaner so I can upload it to github.

I think I will recreate the front end now that it is in a good working state..

Thank You
Leon Melson

closed 2020-08-25 by leonmelson