Klipper3d/klipper #3106: Klipper doesn't seem to listen the M17 command at the end of the G-code

DrieStone, 2020-07-19:
I have an Ender 5, and the last two G-Code commands are:

M84 X Y E ;Disable all steppers but Z
M17 Z E ;Enable Z stepper

So the bed doesn't drop (the Ender 5's bed weight can make the bed drop at the end of a print). Klipper doesn't seem to respect this command and disables the Z stepper when it's done printing.

Gilabite, 2020-07-19:
You can use SET_STEPPER_ENABLE STEPPER=stepper_name ENABLE=0 to disable a specific stepper and ENABLE=1 to enable it in the gcode

Hywelmartin, 2020-07-20:
here's supported G-codes 
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

mental405, 2020-07-24:
M17 is not a supported gcode. 

You could create macros to handle it using the SET_STEPPER_ENABLE commands

`
[gcode_macro M17]
gcode:    
    {% if 'X' in params %}
        SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=1
    {% endif %}
    {% if 'Y' in params %}
        SET_STEPPER_ENABLE STEPPER=stepper_y ENABLE=1
    {% endif %}
    {% if 'Z' in params %}
        SET_STEPPER_ENABLE STEPPER=stepper_z ENABLE=1
    {% endif %}
    {% if 'E' in params %}
        SET_STEPPER_ENABLE STEPPER=extruder ENABLE=1
    {% endif %}    
    {% if   not 'X' in params 
        and not 'Y' in params
        and not 'Z' in params
        and not 'E' in params %}            
        SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=1
        SET_STEPPER_ENABLE STEPPER=stepper_y ENABLE=1
        SET_STEPPER_ENABLE STEPPER=stepper_z ENABLE=1
        SET_STEPPER_ENABLE STEPPER=extruder ENABLE=1
    {% endif %}   
      

[gcode_macro M18]
rename_existing:            M18.1
gcode:   
    {% if 'X' in params %}
        SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=0
    {% endif %}
    {% if  'Y' in params %}
        SET_STEPPER_ENABLE STEPPER=stepper_y ENABLE=0
    {% endif %}
    {% if 'Z' in params %}
        SET_STEPPER_ENABLE STEPPER=stepper_z ENABLE=0
    {% endif %}
    {% if 'E' in params %}
        SET_STEPPER_ENABLE STEPPER=extruder ENABLE=0
    {% endif %}     
    {% if   not 'X' in params 
        and not 'Y' in params
        and not 'Z' in params
        and not 'E' in params %}              
        M84
    {% endif %}   
`

closed 2020-07-26 by klipper-gitissuebot