Klipper3d/klipper #3107: Strange thermals...

Hairyloon, 2020-07-19:
I expect the most probable answer to this is dodgy wiring, but I thought I would ask here anyway in case.
I'm printing with PETG for the first time, I've done a couple of successful prints, but now I've developed a curious problem: see graph.
[image]

Clearly the two peaks are where is has shut down and I've re-started it, but it is the shape of the curve that has caused it to shut down, and the second curve I find particularly odd.
Any thoughts as to what has caused it to do that?

While I am here, I have a couple of questions that are only tangentially related, but I don't think it worth starting a new thread, or if it is, then I will.
Firstly, it would be helpful to sound an audio alarm when it has a crisis and shuts down. Is there a facility in Klipper to do that?
It shouldn't be hard to attach a buzzer or speaker to a spare pin, but I wouldn't know how to trigger it.
Secondly, it takes forever to cancel a print. I suspect that this is an Octoprint problem rather than klipper which is why I ask it as an aside.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4943759/klippy.log)

robthide37, 2020-07-19:
pid tune

ld3300, 2020-07-20:
Yeah, it does look like PID tuning would help.  Did you intentionally overshoot and hit your max temp threshold?

Look in the Beeper section of the sample_macros.cfg for setting up something like a piezo.  However, if Klipper shuts down due to an error or e-stop it will not be running to sound the buzzer.  If using a raspberry pi that could potentially run it for those scenarios.

I believe that one of the things that makes cancelling a print inconsistent is that it won't interrupt a g-code command, but has to wait until it can jump in.  I could be mistaken on that.  

David

ld3300, 2020-07-21:
hmm, now you are completely beyond me.  I haven't had a cancel take that long.

Hairyloon, 2020-07-21:
Curiously enough, I've not had it take so long since I complained about it. I suspect that is coincidence.
Nor have I had a repeat of the thermal event, so I'll close this now, although I would like to understand how the PID tune cured it (assuming that it did), if anybody is minded to try to explain.

closed 2020-07-22 by Hairyloon