Klipper3d/klipper #3109: Alfawise U30 - Failed to flash

s00999-69, 2020-07-19:
Hi guys,

I try to flash my U30 with klipper without success.

I get following message when I execute the following command : 

make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Result : 

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1eaf:0003

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

make: *** [src/stm32/Makefile:55: flash] Error 255

I don't have any klipper log file in /tmp folder

any idea?

thanks

KevinOConnor, 2020-07-19:
Be sure to follow the directions at the top of https://github.com/KevinOConnor/klipper/blob/master/config/printer-alfawise-u30-2018.cfg

-Kevin

s00999-69, 2020-07-20:
ok my mystake. 
It seems to work now.

I receive the temp information from my U30 but when I want to send a command, i get the following result : 

Recv: ok B:28.2 /0.0 T0:28.5 /0.0
Send: STATUS
Recv: // Shutdown due to M112 command
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: // Klipper state: Not ready
Recv: !! Shutdown due to M112 command
Changing monitoring state from "Operational" to "Error: Shutdown due to M112 command"
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Changing monitoring state from "Error: Shutdown due to M112 command" to "Offline (Error: Shutdown due to M112 command)"
Connection closed, closing down monitor

any idea?

thanks

jakep82, 2020-07-20:
You need to post a log file.

s00999-69, 2020-07-21:
Hi,

here is the log file.

thanks
[klippy.log](https://github.com/KevinOConnor/klipper/files/4954854/klippy.log)

s00999-69, 2020-07-23:
Hello guys, any idea?

thansk

KevinOConnor, 2020-07-25:
That's certainly odd.  Try completely powering down the raspberry pi and the alphawise printer.  Then bring them both up.  If you continue to see a problem then post the new log file.

-Kevin

s00999-69, 2020-07-26:
Hi, same issue after shutdown/restart of PI and U30.
Log file is attached.

thanks
[klippy.log](https://github.com/KevinOConnor/klipper/files/4978061/klippy.log)

KevinOConnor, 2020-07-28:
Some software is asking Klipper to go into "emergency shutdown mode" by issuing an M112 command.  If you're using OctoPrint, make sure you've followed all the steps at: https://www.klipper3d.org/Installation.html#configuring-octoprint-to-use-klipper .  If you're using some software other than OctoPrint, you'll need to track down why it's issuing an M112 command.

-Kevin

closed 2020-08-18 by KevinOConnor