Klipper3d/klipper #3112: Extruder turns off after print starts

Printy, 2020-07-21:
### What happens:
About 10 seconds after the printing has started(after nozzle wipe) the extruders target temperature drops to 0 (see log line 489).
This only happens at the start of the print job. The print finishes without any problems if I reset the temperature after the drop.
This problem happens with any printing job.

I have also included a gcode file just in case.

### Setup:
Hardware: Stock Tevo Tornado + RP3B
Software: Klipper/Octoprint w/klipper addon
Slicer: Cura 4.6

[klippy.log](https://github.com/KevinOConnor/klipper/files/4955782/klippy.log)
[TT_xyzCalibration_cube.gcode.txt](https://github.com/KevinOConnor/klipper/files/4955889/TT_xyzCalibration_cube.gcode.txt)

jakep82, 2020-07-22:
You need to remove the M109 R225 line from your gcode. 

https://github.com/KevinOConnor/klipper/issues/2370

Printy, 2020-07-22:
@jakep82 Thank you for the solution!
Any ideas how to make cura not generate that gcode line?

jakep82, 2020-07-22:
Sorry, I don't use Cura and I'm not familiar with it.

closed 2020-07-23 by Printy