Klipper3d/klipper #3114: TMC2209 wiring

PerLangkjaer, 2020-07-22:
Hello all
I have yet another TMC2209 question as i can't seem to find an answer on the interweb.
I have just gotten some Makerbase TMC2209 V2.0
https://github.com/makerbase-mks/MKS-StepStick-Driver/tree/master/MKS%20TMC2209/MKS%20TMC2209%20V2.0_001
But i don't know how to wire them to my Rumba board.
I had some 2208 running, but unfortunatly they don't run anymore.
Will somebody be so kind to guide me to what i need to solder on the drivers and what pins to remove (in any) and what pin for the UART.
It's not that I can't read, but i can't understand the schematics :-(

Best Regards from Sweden

mental405, 2020-07-24:
Looking at the image of that driver on their Ali store, it looks like there is a 0ohm resistor "R8" which should be jumpered to enable UART mode.
[image]

I don't know what board you are connected them to for UART or how you had your 2208's connected.

If it is an SKR 1.3 or something with the integrated UART lines they should be plug and play at that point.

If it is RAMPS or MKS Gen L or something and you had jumpers running from the drivers to an aux pin or something on the board you would need to heat the UART pin up and push it through the to the top of the board. A single wire jumper without a resistor can be used on the same pins as your 2208s.

PerLangkjaer, 2020-07-24:
@mental405 
It's a RUMBA board, not the RUMBA32, but the "old" one.

mental405, 2020-07-24:
Thats what im saying, im am not familiar with it really, but it is probably one where you need to pull all the MS jumpers and run UART wires.

PerLangkjaer, 2020-07-24:
@mental405 
I warmed and pressed the UART pin throu and nothing else. now it works a treat :-D
I did the same with BIGTREETECH V1.1 and it worked as well.
Remember to remove the jumpers to MS1 and MS2 and MS3, I forgot and where a little disappointed.
Remembed it and it worked.
And as I write this I see you comment ;-)
Thanks for the help.

closed 2020-07-24 by PerLangkjaer