Klipper3d/klipper #3115: Missing steps on X axis

PerLangkjaer, 2020-07-23:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4966588/klippy.log)
My printer is missing steps on the X axis
I have tried TMC2208, TMC2209 and DRV8825 drivers. I have tried multiple drivers, and swapped over from Y to X and still the same.
The TMC2208 drivers where in UART and spreadcycle modes and I have tried from 0.6 to 1.2 in RUN_CURRENT.
When I used Marlin 1.1.9 it was never a problem, but i wanted to move this printer to Klipper as well.
I have a RUMBA board with ATmega chip.
Could it be the motor that has something to do with it?

Hywelmartin, 2020-07-23:
sound like physic or mechanical issue

PerLangkjaer, 2020-07-23:
Unfortunately it does not feel like a a mechanical problem, the bed moves freely and as it worked on marlin I'm guessing it could be in the firmware.

jakep82, 2020-07-23:
What type of printer are you using?  The bed on a typical i3 style printer is the Y axis, not the X axis.  I have a printer with a RUMBA board and I have no issues with losing steps.  I've used the board with both TMC2100 and TMC2208 drivers without a problem.

n8bot, 2020-07-23:
Saying it worked on Marlin doesn't really say much. What wee the acceleration settings? Jerk settings? Max speed? Commanded speed? Microstepping setting? All of these (and more) can affect an axis' propensity to skip steps.

PerLangkjaer, 2020-07-23:
Sorry it actually is the Y axis, but its a Velleman K8200 and the bed is moving in both X and Y

I have 500 i acc and between 20 and 50 mm/s for moves with 120 i travel.

It is not a mechanical problem, I have taken apart the bed gantry, cleaned everything and put back together. everything lines up and there are no place where it needs some force to move the bed.
Right now I'm swithing to another motor, just to try something.
next I'll have to try another board if this do not work.

PerLangkjaer, 2020-07-24:
It looks like the motor wasn't really happy with how Klipper wanted to use it.
I did a small test print in the night and it went well.
I'm going to close this now as I believe it's fixed.
Now I just need to figure out how to wire my 2209 to my board ;)

closed 2020-07-24 by PerLangkjaer