Klipper3d/klipper #3116: pressure_advance cause mcu shutdown

Daeamon, 2020-07-23:
Hello, mcu shutdown while printing can you help me to solve this problem? I want to use pressure_advance option but it cause mcu shutdown.((( The firmware control retraction i set speed to 20 and retract length to 0.75 but it does not help. After the first layer, the mcu is shutdowned.

[CFFFP_hinge.gcode.txt](https://github.com/KevinOConnor/klipper/files/4967806/CFFFP_hinge.gcode.txt)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4967795/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4967789/klippy.log)

Sineos, 2020-07-23:
MCU saturated by 64 microsteps of the extruder?
Have you tried lowering it to 32 and maybe turn on interpolate instead?

Daeamon, 2020-07-23:
Thanks, this help for me.

closed 2020-07-23 by Daeamon