Klipper3d/klipper #3119: Klipper Sensorless Homing not working - SKR v1.3 - TMC2209 - CoreXY

furkanbr, 2020-07-23:
Hi everyone , im working on my new corexy machine and want to try klipper firmware but sensorless homing not working on my machine. i tried everythink that i find but still same , toolhead hitting to mechanical limit but still want to move not sensing it. Please help me i cant make it work. im using RPi 4 4gb , skr v1.3 , tmc2209 stepper driver made by fysetc and nema 17 42 mm stepper motor. İm waitting for your answers , thanks , have a good day.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4968716/klippy.log)

BlackStump, 2020-07-23:
Have you been through the [sensorless homing document](https://www.klipper3d.org/Sensorless_Homing.html)

furkanbr, 2020-07-23:
> Have you been through the [sensorless homing document](https://www.klipper3d.org/Sensorless_Homing.html)

yes and tried other solutions in githup issues , i didnt find what can be a problem. im using this configuration for x axis but not working ;

[stepper_x]
step_pin = P2.2
dir_pin = P2.6
enable_pin = !P2.1
step_distance = .0063
endstop_pin = tmc2209_stepper_x:virtual_endstop
position_endstop = 220
position_max = 220
homing_speed = 50
homing_retract_dist = 0

[tmc2209 stepper_x]
uart_pin = P1.17
diag_pin = P1.29
microsteps = 16
run_current = 0.7
hold_current = 0.4
stealthchop_threshold = 200
driver_sgthrs = 200

BlackStump, 2020-07-23:
I personally have no experience on sensorless homing, personally I do not see the attraction over a mechanical switch but having said that afaik sensorless homing works fine with klipper.
That would suggest the places to research is the skr1.3 implementation and the flysetc as it would seem you have a hardware issue not a klipper firmware issue.
As you have probably seen in other issues the best klipper advice is the doc I linked for you, maybe someone with with experience with your particular hardware will chime in.

furkanbr, 2020-07-23:
Yes i know i saw few video on youtube that works but non of them not shared their cfg files. i changed now my drivers with not used ones with driver_sgthrs = 255 (max sens) but nothing changed , still same. print core hit to y axis carriage and still trying to move. 

when i checked my endstops with QUERY_ENDSTOPS , i can see all of them open. 

But there is somethink i didnt understand , when i check my x stepper driver with DUMP_TMC stepper=stepper_x,  that gives me different answer than "Sensorless homing" guide "Testing of SPI/UART communication" part ;

`Send: DUMP_TMC stepper=stepper_x

Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 0008160c IHOLD=12 IRUN=22 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000013 TPWMTHRS=19
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // SGTHRS:     000000ff SGTHRS=255
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c4 en_spreadCycle=1 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      00000018 IFCNT=24
Recv: // OTP_READ:   0000000c OTP_FCLKTRIM=12
Recv: // IOIN:       21000141 ENN=1 PDN_UART=1 SPREAD_EN=1 VERSION=0x21
Recv: // FACTORY_CONF: 0000000c FCLKTRIM=12
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      00000128 MSCNT=296
Recv: // MSCURACT:   01c300ef CUR_A=239 CUR_B=-61
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: 800c0000 CS_ACTUAL=12 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  0000000e PWM_SCALE_SUM=14
Recv: // PWM_AUTO:   000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
Recv: // SG_RESULT:  00000008 SG_RESULT=8
Recv: ok`

what these mean for me?

thomask77, 2020-07-24:
Don't know if it helps, but here's a working config for a CoreXY running on a SKR Mini 2.0.
Also TMC2209 with sensorless homing. On the SKR Mini, you would have to set the DIAG jumpers for X and Y axis.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4970961/printer.cfg.txt)

mental405, 2020-07-24:
I presently have sensorless working on a Voron with 2209s.

The driver_SGTHRS parameter for these drivers goes from 0-254 iirc where 0 is least sensitive and 254 is most sensitive.

The 2209s will only home in stealthchop so that parameter should have no bearing on it.

The only thing I can suggest is perhaps increase your SGTHR up to 254 and if it still does not trigger, try upping the homing speed a bit. I home at 60.

Another thing is that stallguard works best when it gets a running start, so make sure the toolhead is off the axis before you try homing.

thomask77, 2020-07-24:
"The 2209s will only home in stealthchop" - I have StealthChop deactivated completely, and it still works.

Would you like to share your config for reference?

furkanbr, 2020-07-25:
@thomask77 Thanks , i tried to configure like your .cfg file , nothink changed , eved tried to dissable stealthchop with "#" like you did but didnt make any differance. 

@mental405 Thanks , i tried everytink you said , i did my all tests at driver_SGTHRS = 254 , and speed up my hpming speed from 30 to 100 mm/s , all of my motors are off before homing test and increased my motor current from 0.7&0.5 to 1.7&1.2 amp ( my max current for motors is 1.7 amp) ,x,y,z of my sensorless homing jumpers are connected , i double checked my fysets 2209 drivers pinout is same with btt 2209 driver , im using this board , drivers and motors first time , endstop pins empty , i tested homing with 120 ,250 and even dissabled (with "#") stealthchop but no one of this configucarions did not work.

i dont know what am i doing wrong , do anyone have any ideas? Thanks to all

furkanbr, 2020-07-25:
All of my connections for now 
[image]

mental405, 2020-07-25:
@thomask77 I don't think this is accurate. I am pretty sure for sensorless to work they have to be configured in UART mode. The code that actually performs the homing action turns stealthchop on for the homing move and then sets it back to the previous setting once it has completed.

```
   def handle_homing_move_begin(self, endstops):
        if self.mcu_endstop not in endstops:
            return
        reg = self.fields.lookup_register("en_pwm_mode", None)
        if reg is None:
            # On "stallguard4" drivers, "stealthchop" must be enabled
            self.mcu_tmc.set_register("TPWMTHRS", 0)
            val = self.fields.set_field("en_spreadCycle", 0)
        else:
            # On earlier drivers, "stealthchop" must be disabled
            self.fields.set_field("en_pwm_mode", 0)
            val = self.fields.set_field("diag1_stall", 1)
        self.mcu_tmc.set_register("GCONF", val)
        self.mcu_tmc.set_register("TCOOLTHRS", 0xfffff)
```

@furkanbr your board looks correct. The UART jumpers appear to be set correctly for the axes in question and the sensorless jumpers are also set correctly. This is probably not the issue but the wiring going into the board looks a little thin. Hopefully this is just set up that way for testing because it could melt it when you start driving a hotend and bed. This is what I have for my X motor on my CoreXY. My motors are 2.0A motors with a 1.8 degree step angle. The pin numbers might be different because I was routing based on where my wires would reach. This might be a silly question but does the diag pin come out both sides of that stepper driver? My old TMC 2130s only had the diag pins coming out the top. The red 2 pin header that is perpendicular is where it should go. and the pins should be the same length on the underside as the rest of the pins.

```
[stepper_x]
step_pin:                   P0.19
dir_pin:                    P0.20
enable_pin:                 !P2.8
endstop_pin:                tmc2209_stepper_x:virtual_endstop 
step_distance:              0.0125
position_min:               0
position_endstop:           300
position_max:               300
homing_speed:               60
homing_retract_dist:        0
homing_positive_dir:        true

[tmc2209 stepper_x] 
uart_pin:                   P1.15
diag_pin:                   P1.27
microsteps:                 16
interpolate:                True
run_current:                1.5
hold_current:               .5
sense_resistor:             0.11
stealthchop_threshold:      1
driver_SGTHRS:              140
```

The pin associations in your config appear to be correct. 
```
Socket  UART  DIAG  STEP  DIR  EN
X       1.17  1.29  2.2   2.6  2.1
Y       1.15  1.27  0.19  0.20 2.8
Z       1.10  1.25  0.22  2.11 0.21
E0      1.8   1.28  2.13  0.11 2.12
E1      1.1   1.26  0.1   0.0  0.10
```

The dump TMC report also looks like the UART is working correctly. I am unsure if there is any configuration that needs to be done with the actual stepper drivers. If you have access to an oscilloscope, arduino, analog multimeter, or even a single LED, you could attach it to the diag pin to see if it is outputting a signal when stallguard is active. On stall detection, that pin will output +3.3V according to the datasheet. You should be able to capture that pulse with some sort of diagnostic setup. Ideally that setup would be the scope but an arduino dumping the contents of analogRead() into a serial graph would capture it. I sufficient digital multimeter might work as well but mine samples too slowly to be up to the task. The last resort is the LED, it should provide a visual indicator of when the stall occurs but it may just be a quick flash.... idk, i have never tried it.

> 
> DIAG 11 DO
> Diagnostic and StallGuard output. Hi level upon stall detection or
> driver error. Reset error condition by ENN=high.

furkanbr, 2020-07-25:
@mental405 thanks for your answer , yes this setup just for testing my x , y axis and sensorless homing. im using fysetc 2209 v3 model and yes diag pin and 3 others go all the way up and down , they are connected to board. i tried your configurations with right pins and nothing changed , then i tried your test solution , with multimeter and LED. i connected an LED to DIAG and GND pin but when i try to home , LED didnt light on , tried few other configuration but didnt changed anythink , LED at DIAN pin is never turned on. 

i have 5 tmc2209 driver and i didnt use them ever before , non of them didnt work. What can be problem about that? Thanks a lot.

furkanbr, 2020-07-26:
@thomask77  @mental405 @BlackStump @pusewicz 
For new update , its still not working but i have another problem and it can be because of homing problem. 

i was using my tmc2209 drivers on skr 1.3 with uart and using 1 jumper bottom of driver as you can see of the photo with z and e axis. i was connected that jumper because klipper giving me an error and someone at reddit said that as a solution then i tried and error is gone. But at skr 1.3 docs , for uart mode , we sould unconnect all jumpers from bottom of driver. so i think about that and want to try , i delete all unneccessary parts from my configuration and remove all jumpers from bottom of jumbers then tried for homing. But my problem start at here , tmc drivers giving an error , they are not connecting to skr , and i cant even try homing. Skr says remove all jumpers but if i remove , its giving me an error. i tried 5 brand new driver to be sure but all of them same. what should i do not , what am i doing wrong ? Thanks to everyone

My board to see old jumper connection 

[image]

New Error:
[image]

My Latest Configuration :
```
[stepper_x]
step_pin:                   P2.2
dir_pin:                    P2.6
enable_pin:                 !P2.1
endstop_pin:                tmc2209_stepper_x:virtual_endstop 
step_distance:              0.0125
position_min:               0
position_endstop:           300
position_max:               300
homing_speed:               60
homing_retract_dist:        0
homing_positive_dir:        true

[tmc2209 stepper_x] 
uart_pin:                   P1.17
diag_pin:                   P1.29
microsteps:                 16
interpolate:                True
run_current:                0.8
hold_current:               .5
sense_resistor:             0.11
stealthchop_threshold:      1
driver_SGTHRS:              254

[stepper_y]
step_pin:                   P0.19
dir_pin:                    P0.20
enable_pin:                 !P2.8
endstop_pin:                tmc2209_stepper_y:virtual_endstop 
step_distance:              0.0125
position_min:               0
position_endstop:           300
position_max:               300
homing_speed:               60
homing_retract_dist:        0
homing_positive_dir:        true

[tmc2209 stepper_y] 
uart_pin:                   P1.15
diag_pin:                   P1.27
microsteps:                 16
interpolate:                True
run_current:                0.8
hold_current:               .5
sense_resistor:             0.11
stealthchop_threshold:      1
driver_SGTHRS:              254

[stepper_z]
step_pin: P0.22
dir_pin: P2.11
enable_pin: !P0.21
step_distance: .01     # 4 (calculated) 
step_distance: .0025     # 16 (calculated)
position_endstop:           300
endstop_pin: P1.25
position_max: 400
homing_speed: 30
 
#[tmc2209 stepper_z]
#uart_pin: P1.10
#microsteps: 16
#interpolate: True
#run_current: 1.27
#hold_current: 0.9
#stealthchop_threshold: 183 # 275 RPM
#stealthchop_threshold: 60

[safe_z_home]
home_xy_position: 160,160
speed: 700
z_hop: 10
z_hop_speed: 100

[thermistor mosquito_thermistor]
temperature1 = 20
resistance1 = 613400
temperature2 = 190
resistance2 = 4509
temperature3 = 260
resistance3 = 1359

[extruder]
max_extrude_only_distance: 100.0
step_pin: P2.13
dir_pin: !P0.11
enable_pin: !P2.12
nozzle_diameter: 0.400
filament_diameter: 1.75
heater_pin: P2.7
sensor_type: mosquito_thermistor
sensor_pin: P0.24
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 270
step_distance: 0.002409  # (BMG recommended value)
max_extrude_cross_section: 5

[tmc2209 extruder]
uart_pin: P1.8
microsteps: 16
interpolate: False
run_current: 1.07
hold_current: 0.869
#stealthchop_threshold: 183 # 275 RPM
stealthchop_threshold: 10

[mcu]
serial: /dev//ttyACM0

[printer]
kinematics: corexy
max_velocity: 500
max_accel: 4500
#max_accel_to_decel: 400
max_z_velocity: 30
max_z_accel: 100
#square_corner_velocity: 3
```

BlackStump, 2020-07-26:
I think you need to check if the fysetc 2209's are compatible out of the box
[fysetc](https://wiki.fysetc.com/Silent2209/)
[bigtreetech](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/manual/TMC2209-V1.2-manual.pdf)
[bigtreetech-problems](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf)

furkanbr, 2020-07-26:
@BlackStump @pusewicz @mental405 @thomask77 @KevinOConnor 

Yes i saw few other guy using fysetc 2209 driver , for tmc uart error "Unable to read tmc uart 'stepper_x' register IFCNT" , im solving that error like this guy said ; 

[image]

with this setup , i dont get any tmc uart error but my drivers working for homing. when i try to sensorless home , axis hitting to y carriage and try to still move , LED on DİAG-GND pins not turning on when im trying to home. i still cant solve the problem , even dont know what is the problem. 

My DUMP_TMC stepper=stepper_x :
```
Send: DUMP_TMC stepper=stepper_x
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 0008190f IHOLD=15 IRUN=25 IHOLDDELAY=8
Recv: // TPWMTHRS:   0000004f TPWMTHRS=79
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // SGTHRS:     000000fe SGTHRS=254
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c4 en_spreadCycle=1 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      0000001b IFCNT=27
Recv: // OTP_READ:   0000000d OTP_FCLKTRIM=13
Recv: // IOIN:       21000141 ENN=1 PDN_UART=1 SPREAD_EN=1 VERSION=0x21
Recv: // FACTORY_CONF: 0000000d FCLKTRIM=13
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      000000b8 MSCNT=184
Recv: // MSCURACT:   006800e0 CUR_A=224 CUR_B=104
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: 800f0000 CS_ACTUAL=15 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  00000012 PWM_SCALE_SUM=18
Recv: // PWM_AUTO:   000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
Recv: // SG_RESULT:  00000014 SG_RESULT=20
Recv: ok
```
DUMP_TMC looking different then klipper docs but is this a problem?

My current klippy log file :

[klippy.log](https://github.com/KevinOConnor/klipper/files/4977683/klippy.log)

furkanbr, 2020-07-28:
@thomask77 @mental405 @BlackStump @pusewicz @KevinOConnor 

For to be sure , i installed Marlin 2.0 to my skr 1.3 , removed all jumpers , reconnect 7 of them for uart connection and sensorless homing for all axes then i start my printer , checked with pronterface. All of my 4 tmc drivers gave me a connection error ( ALL LOW) , then i tried again with 1 jumper connection at below all stepper drivers like i tried before at Klipper. Error is gone , i configured all driver settings like current , sensivity. Then i tried for sensorless homing but same think happend with Klipper , motor still trying to move. 

My setup cant home with tmc drivers with 2 different firmware , what should i do now?

Can you guys help me for this problem? my all electronics are brand new , i didnt use them before , what could i have done wrong? 

Thanks too everyone!

davidsteinberger, 2020-07-29:
Can u double check if you are truly running stealthChop. According to your _ DRV_STATUS_ it does not look like.

Sineos, 2020-07-29:
Have you tuned the driver's voltage via the potentiometer?

I cannot confirm myself but read reports about these driver misbehaving when current is controlled via UART but voltage potentiometer no set.

furkanbr, 2020-07-29:
@davidsteinberger  Thanks , i realized that after then you said , my DRV_STATUS (CS_ACTUAL=15 stst=1) dont have any stealthChop value and i saw someone has " CS_ACTUAL=31 stealth=1 stst=1" value in DRV_STATUS. My stealthchop_threshold: 1 and i tried with 250 too , did i miss anythink about stealthChop mode? have can i activate it ?

@Sineos Yes , at my first tests , i tuned my potentiometer on tmc for step/dir mode , its giving me 0.8 v vref value for now , should i change that ? 

Thanks to all !!

furkanbr, 2020-07-29:
@thomask77 @davidsteinberger @Sineos @mental405 @BlackStump @KevinOConnor 

Alot of people saying "stallguard not working on stealthchop mode" , then edit my .cfg file , set my stealthchop_threshold: 0 , shortly i dissabled it. want to check my DUMP_TMC and that give me ;

```
Send: DUMP_TMC stepper=stepper_x
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081711 IHOLD=17 IRUN=23 IHOLDDELAY=8
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // SGTHRS:     000000fe SGTHRS=254
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      0000002d IFCNT=45
Recv: // OTP_READ:   0000000c OTP_FCLKTRIM=12
Recv: // IOIN:       21000141 ENN=1 PDN_UART=1 SPREAD_EN=1 VERSION=0x21
Recv: // FACTORY_CONF: 0000000c FCLKTRIM=12
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      00000288 MSCNT=648
Recv: // MSCURACT:   015a0148 CUR_A=-184 CUR_B=-166
Recv: // CHOPCONF:   14010053 toff=3 hstrt=5 TBL=2 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0110000 CS_ACTUAL=17 stealth=1 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  001400a3 PWM_SCALE_SUM=163 PWM_SCALE_AUTO=20
Recv: // PWM_AUTO:   000e00ff PWM_OFS_AUTO=255 PWM_GRAD_AUTO=14
Recv: // SG_RESULT:  00000010 SG_RESULT=16
Recv: ok
```
in DRV_STATUS line , i have stealth=1 value now , then i tried for homing again , bt nothing changed. Still same , stallguard not triggared. 

am i making somethink wrong? what am i missing?

KevinOConnor, 2020-08-04:
If I understand you correctly, you are indicating that sensorless homing isn't working in Marlin nor Klipper.  This sounds like the "diag pin" isn't being routed from the tmc2209 chip to the configured pin on the micro-controller.  Some of the step-sticks and control boards have goofy wiring, so it could be a problem there.  Beyond that I'm not sure there's much assistance we can offer.

-Kevin

furkanbr, 2020-08-04:
@KevinOConnor  Thanks , im gonna check that maybe i can solve , i ordered an skr mini v2 board if i cant figure my issue with skr 1.3 im gonna use that but first im gonna check my pin connections.

slavedas, 2020-08-06:
@furkanbr I had a similar problem with the Eryone 2209s. It turns out the Diag1 pin was actually NEXT to where it is supposed to be.  I had to run a flying wire from the top pin to the bottom of the DIAG jumper location (Take out the diag jumper) to solve it. THAT worked.

So on your stepper, on the far left bottom, there are two pins, the top on SHOULD be DIAG1, but on mine, it was DIAG0 and DIAG1 was the bottom of the two pins.

furkanbr, 2020-08-14:
@thomask77 @davidsteinberger @Sineos @mental405 @BlackStump @KevinOConnor

My new board SKR Mini E3 v2.0 is arrived and i try that board with marlin , everythink and sensorless homing working fine now , im gonna try Klipper with this board if i dont get any issue with marlin , just for to be sure all of my hardware is okay. 

@slavedas i didnt check skr 1.3 again but im gonna check and if i can make it work , i can use that with my other printer , thanks!

kakou-fr, 2020-08-18:
I use an SKR1.3, BTT TMC2209 on a core-xy and sensorless homing works 

```
[tmc2209 stepper_x]
uart_pin: P1.17
microsteps: 32
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
interpolate: True
sense_resistor: 0.110
diag_pin: ^P1.29
driver_SGTHRS: 100

[tmc2209 stepper_y]
uart_pin: P1.15
microsteps: 32
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
interpolate: True
sense_resistor: 0.110
diag_pin: ^P1.27
driver_SGTHRS: 100

[stepper_x]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
step_distance: .00625
#endstop_pin: P1.29
position_endstop: -1
position_min: -1
position_max: 375
homing_speed: 100
homing_retract_dist: 0
endstop_pin: tmc2209_stepper_x:virtual_endstop

[stepper_y]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
step_distance: .00625
#endstop_pin: P1.27
position_endstop: 380
position_min: -20
position_max: 380
homing_speed: 100
homing_retract_dist: 0
endstop_pin: tmc2209_stepper_y:virtual_endstop
#homing_positive_dir: true
```

furkanbr, 2020-08-19:
@kakou-fr Yes i saw fex guys using skr 1.3 and btt 2209 , maybe the problem is not board , i was using fystec 2209 drivers and saw another guy with same setup and issue , havent tried again yet but i tried everythink i can it didnt work. 

For now im working on SKR Mini v2.0 with marlin fw but im not that much happy because im looking for high speed printing and at current setup its skipping step even at 4000mm/s'2 , thats why i want to change that to klipper.

@thomask77 My friend you said you are using SKR mini v2.0 , i just tried to install Klipper with my Rpi 4 and SKR mini v2 but i failed at flashing firmware to microcontroller. " ls /dev/serial/by-id/* " command not giving me any port and then i cant even connect to my board. i tried to copy klipper.bin file to sd card of board for flashing it but it didt work complately. 

it respon me but giving an error :
```
Send: status
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: // Klipper state: Not ready
Recv: !! Unable to connect
Recv: ok
```
How did you make it work , did i miss anythink? Thanks to everyone!

BlackStump, 2020-08-19:
I presume you are talking about the SKR Mini E3 V2.0? if so did you follow the flashing details on the top of the [SKR Mini E3 Config](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg)
or did you flash for the 2KB [HID Bootloader](https://github.com/KevinOConnor/klipper/blob/master/docs/Bootloaders.md#stm32f103-with-hid-bootloader) ?
If you did flash for the HID Bootloader then you will follow through with the HID Bootloader instructions

furkanbr, 2020-08-21:
@BlackStump Thanks you!!! You make me so happy my friend :D i wasnt checked config file and missing "Also, select "Enable extra # low-level configuration options"" part on my setup , i deleted everythink and restart again , followed all steps on SKR Mini E3 Config and it works now , and tried to sensorless home and it works too!! the only part that i didnt understand  on installing is " ls /dev/serial/by-id/* " command not giving any port value on return , my mcu adress is " /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFD8054247343133660957-if00 " i write that to printer.cfg file and it works , i think its okay but i didnt understand why i it like that.

My problem is solved by @BlackStump  (thanks again) and my steps ;
*SKR v1.3 + Fsyetc 2209 setup still not working , i dont know why , it can be board or driver issue , most cammon 2209 driver is from BTT and maybe they can be different in some point , they works on STEP/DİR mode but not working on uart mode.
*I get new SKR Mini v2.0 board with integred 2209 drivers and everythink is working good now.

Thanks to everyone help me about my issue !

closed 2020-09-19 by github-actions[bot]

JAdams12, 2021-07-05:
@slavedas 

Can you expand on your Eryone stallguard solution? I can't find any information on what to do to get it working. So far you're one of the only people who have it working that I can find

BlackStump, 2021-07-05:
The info has been provided,
but even simpler if you have a BTT Board use BTT TMC2209 and if you have FYSETC Board use FYSETC TMC2209
BTT is pinned out differently to FYSETC if you mix and match you have to figure out the differences.

JAdams12, 2021-07-05:
> The info has been provided,
> 
> but even simpler if you have a BTT Board use BTT TMC2209 and if you have FYSETC Board use FYSETC TMC2209
> 
> BTT is pinned out differently to FYSETC if you mix and match you have to figure out the differences. 

I was referring to the user who said they had Eryone drivers working. They said the diag pin is next to where it should be, I'm not sure what that means exactly and was hoping to get a picture or some extra information about it. I couldn't figure out how to message them on GitHub and ask.

There's a lot of information out there on the Fysetc and BTT drivers but there's next to nothing on the Eryone ones

BlackStump, 2021-07-06:
arh sorry didn't know Eryone  was another stepper driver supplier with no pinouts for there products

BlackStump, 2021-07-06:
quick search it seems to be the Vref, Index and Diag pins are in different locations
[image]
[image]
[image]

JAdams12, 2021-07-06:
Good find. I wonder which Pin is the one populated. As far as I know it's supposed to be the diag pin connected to the endstop port on the board

slavedas, 2021-07-06:
I used the pin on top next to the pin that should be diag1 but it's
actually diag0. I had to use flying wires.