Klipper3d/klipper #3120: E steps broken

Diana6969, 2020-07-24:
Hi everyone, please haaalp
I bought a new extruder, because of under-extrusion, but I have found out that the problem is probably in e steps or in sw in general.
I had e-steps configured to about 0.046 or something like that and when I tried to calibrate the printed and tried to extrude 100 mm of filament, I got only 23mm. I calculated the new value, which should be about 2 steps distance in mm.
The problem is the extruder makes weird sounds, while clearly working, but actually moving the filament by MICRONS, or really low distances...
PLEASE HELPPPP
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4973152/printer.cfg.txt)

jakep82, 2020-07-24:
> I calculated the new value, which should be about 2 steps distance in mm.

I'm not sure what this means.  Your steps calculation is below based on the information you provided.

`(23/100)*.046=.01058`

Diana6969, 2020-07-26:
OK, I was using a bad formula from google and from some guy on Youtube, which just didn't work or I made some mistake in this stupidly simple math...
Your formula worked I just had to do the calibration 3 times to get it to perfection! 
Thanks a lot Jakep82

closed 2020-07-26 by Diana6969