Klipper3d/klipper #3121: After pause and raise Z axis, move out of range

darkpoetry, 2020-07-24:
Klipper will fail at the end of a layer with "move out of range" if at some point during that layer you pause the print, raise the extruder using octoprint's control tab by 10mm, then lower it back, then resume.

The only other issues I found relate to bed leveling which I don't use. Many thanks for your work and assistance!

[klippy-pla.log](https://github.com/KevinOConnor/klipper/files/4974114/klippy-pla.log)

Hywelmartin, 2020-07-24:
https://github.com/KevinOConnor/klipper/blob/5f2cb5436cf545c2d5efba7994f458b98644ef6f/config/example-extras.cfg#L482
is missing in your config as a start..

darkpoetry, 2020-07-25:
I don't understand why it would need to capture a position if it's resuming at the exact same position. Does pausing cause it to forget where it's at or does the move during pause cause it to go into an assumption of a not safe state? If the latter, why then does it complete the rest of the layer after

Hywelmartin, 2020-07-26:
relative coordinate
Read 8667.416065: 'N23152 G0 X152.073 Y93.085*26\n'
Read 8667.417046: 'N23153 G0 X175.7 Y83.09*40\n'
Read 8667.629832: 'N23154 G0 X190.603 Y32.145*21\n'
Read 8667.630946: 'N23155 G0 X191.666 Y32.074*21\n'
Read 8667.631872: 'N23156 G0 X191.703 Y30.492*26\n'
Read 8667.632809: 'N23157 M117 Color changing...*86\n'
Read 8667.633664: 'N23158 G92 E0*74\n'
Read 8667.634518: 'N23159 G1 E-2.5 F3000*32\n'
Read 8667.635457: 'N23160 G92 E0*65\n'
Read 8667.636265: 'N23161 G91*22\n'               <---------------- here
Read 8667.637062: 'N23162 G1 X+60 Y-60 F6000*75\n' then try to go to X 251.-ish and Y-30-ish( both out of range)
Read 8667.643896: 'M112\n'

darkpoetry, 2020-07-27:
So, the initial answer above ([pause_resume]) led to the solution, but I thought I'd put a few words here in case someone else gets similarly confused. Tl;dr is Don't use the Octoprint Resume Button.

All this was in an effort to implement M600 for filament changes in Klipper. What would happen previously using the https://github.com/KevinOConnor/klipper/blob/master/config/sample-macros.cfg is M600 would pause the print, eject filament, but then resume printing without descending back to the point it left off.

It was not immediately apparent to me that one actually has to type the word RESUME into the terminal to activate the proper macro, NOT select the red "Resume" button on Octoprint. But that's what needs to be don.

closed 2020-07-27 by darkpoetry