Klipper3d/klipper #3123: 686 seg fault: scripts/install-debian.sh

The-Monkey-King, 2020-07-25:
It appears to stop at line 31 of the installation script:
    sudo apt-get install --yes ${PKGLIST}

The last two lines of the report status show: 
```
Processing triggers for libc-bin (2.28-10) ...
./klipper/scripts/install-debian.sh: line 12:   687 Segmentation fault      sudo apt-get install --yes ${PKGLIST}
```
I do have the /etc/systemd/sytem directory. I have sudo privileges. I already installed OctoPrint on this Debian instance and is running nominally.

I tried running the klipper debian installation script manually and encountered the error:
```
import greenlet
Importerror: No module named greenlet

```
After that I gave up.
[debian-install.txt](https://github.com/KevinOConnor/klipper/files/4976246/debian-install.txt)

BlackStump, 2020-07-25:
I use that script on a beaglebone black and a orangepi zero+ without issue

The-Monkey-King, 2020-07-25:
> 
> 
> I use that script on a beaglebone black and a orangepi zero+ without issue

Oooookay. And this helps how? I am not installing on either one of your hardware.

BlackStump, 2020-07-26:
I was only suggesting it works fine on supported hardware. 
As per [FAQ](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3)
Excerpt from FAQ
>Klipper has been run on other machines. The Klipper host software only requires Python running on a Linux (or similar) computer. However, if you wish to run it on a different machine you will need Linux admin knowledge to install the system prerequisites for that particular machine.

Sineos, 2020-07-26:
Judging from the log the apt installation process is running and seems to segfault for whatever reason at the end. Is there anything in the syslog?

Did you try running the apt process manually if it runs through, e.g.:

`sudo apt-get install --yes  avr-libc avrdude binutils-arm-none-eabi binutils-avr gcc-arm-none-eabi gcc-avr libffi-dev libftdi1 libhidapi-libusb0 libncurses-dev libnewlib-arm-none-eabi libnewlib-dev libstdc++-arm-none-eabi-newlib libusb-1.0-0-dev libusb-1.0-doc libusb-dev stm32flash`

Maybe some packages where left behind being installed only half. Try running:

`dpkg-reconfigure -a`

KevinOConnor, 2020-07-28:
Generally, if a system utility like "apt-get" crashes, it indicates the operating system is corrupted (often a bad sd card).  Generally, the solution is to do a full reinstall of the OS.

-Kevin

The-Monkey-King, 2020-08-03:
Sorry, I had life interrupt. Getting back to perform re-test today, hopefully before I need to prepare for Isaias.

The-Monkey-King, 2020-08-03:
Cannot reproduce this error consistently. (4 out of 10 test trials) 
I think it has to do with my Internet connection and/or storage (M2 card - PC) causing corrupting in my downloaded DEB files.
Closing this defect as not a defect.

closed 2020-08-03 by The-Monkey-King