Klipper3d/klipper #3124: Question: OLED Display Screensaver

garrettwp, 2020-07-25:
With OLED displays being supported, is there an option to have the display turn off after n amount of time to prevent screen burn in? If there is not an option, can one be implemented? We do something similar for the QMK keyboard firmware to prevent the display from burning in or shortening it's life span.

KevinOConnor, 2020-07-28:
There is currently no support for disabling the lcd style screens.  If there is a desire to implement it, then an interested developer will need to write the code and submit it.

Cheers,
-Kevin

KevinOConnor, 2020-08-18:
I'm going to close this as it looks like the original question was answered (it is not currently supported).

-Kevin

closed 2020-08-18 by KevinOConnor