Klipper3d/klipper #3125: BLTouch HS mode increasing offsets without intermediate homing

1N4148, 2020-07-26:
Hi,

tried to test "stow_on_each_sample = False", basically it works, but gives only accurate results after a preceding Z-Homing. If I do repeatedly PROBE_ACCURACY or BED_MESH_CALIBRATE METHOD=automatic I'm getting subsequent systematic offsets in measurements. The Z-height 2.90 is the correct one for my BLTouch-nozzle combo.

With "stow_on_each_sample = True" everything is fine.

`Send: G28 Z0
Recv: ok
Send: G90
Recv: ok
[...]
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:162.000 Y:125.000 Z:10.000 (samples=10 retract=2.000 speed=5.0 lift_speed=10.0)
Recv: // probe at 162.000,125.000 is z=2.895000
Recv: // probe at 162.000,125.000 is z=2.900000
Recv: // probe at 162.000,125.000 is z=2.900000
Recv: // probe at 162.000,125.000 is z=2.897500
Recv: // probe at 162.000,125.000 is z=2.895000
Recv: // probe at 162.000,125.000 is z=2.897500
Recv: // probe at 162.000,125.000 is z=2.900000
Recv: // probe at 162.000,125.000 is z=2.897500
Recv: // probe at 162.000,125.000 is z=2.897500
Recv: // probe at 162.000,125.000 is z=2.897500
Recv: // probe accuracy results: maximum 2.900000, minimum 2.895000, range 0.005000, average 2.897750, median 2.897500, standard deviation 0.001750
Recv: ok
[...]
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:162.000 Y:125.000 Z:4.898 (samples=10 retract=2.000 speed=5.0 lift_speed=10.0)
Recv: // probe at 162.000,125.000 is z=3.442500
Recv: // probe at 162.000,125.000 is z=3.440000
Recv: // probe at 162.000,125.000 is z=3.442500
Recv: // probe at 162.000,125.000 is z=3.440000
Recv: // probe at 162.000,125.000 is z=3.442500
Recv: // probe at 162.000,125.000 is z=3.440000
Recv: // probe at 162.000,125.000 is z=3.440000
Recv: // probe at 162.000,125.000 is z=3.442500
Recv: // probe at 162.000,125.000 is z=3.440000
Recv: // probe at 162.000,125.000 is z=3.432500
Recv: // probe accuracy results: maximum 3.442500, minimum 3.432500, range 0.010000, average 3.440250, median 3.440000, standard deviation 0.002839
Recv: ok
[...]
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:162.000 Y:125.000 Z:5.432 (samples=10 retract=2.000 speed=5.0 lift_speed=10.0)
Recv: // probe at 162.000,125.000 is z=3.985000
Recv: // probe at 162.000,125.000 is z=3.985000
Recv: // probe at 162.000,125.000 is z=3.990000
Recv: // probe at 162.000,125.000 is z=3.987500
Recv: // probe at 162.000,125.000 is z=3.987500
Recv: // probe at 162.000,125.000 is z=3.987500
Recv: // probe at 162.000,125.000 is z=3.987500
Recv: // probe at 162.000,125.000 is z=3.987500
Recv: // probe at 162.000,125.000 is z=3.990000
Recv: // probe at 162.000,125.000 is z=3.987500
Recv: // probe accuracy results: maximum 3.990000, minimum 3.985000, range 0.005000, average 3.987500, median 3.987500, standard deviation 0.001581
Recv: ok
[...]
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:162.000 Y:125.000 Z:5.987 (samples=10 retract=2.000 speed=5.0 lift_speed=10.0)
Recv: // probe at 162.000,125.000 is z=4.525000
Recv: // probe at 162.000,125.000 is z=4.522500
Recv: // probe at 162.000,125.000 is z=4.527500
Recv: // probe at 162.000,125.000 is z=4.522500
Recv: // probe at 162.000,125.000 is z=4.522500
Recv: // probe at 162.000,125.000 is z=4.522500
Recv: // probe at 162.000,125.000 is z=4.522500
Recv: // probe at 162.000,125.000 is z=4.522500
Recv: // probe at 162.000,125.000 is z=4.525000
Recv: // probe at 162.000,125.000 is z=4.520000
Recv: // probe accuracy results: maximum 4.527500, minimum 4.520000, range 0.007500, average 4.523250, median 4.522500, standard deviation 0.001953
Recv: ok`

[klippy.log](https://github.com/KevinOConnor/klipper/files/4977524/klippy.log)
[terminal.log](https://github.com/KevinOConnor/klipper/files/4977525/terminal.log)

KevinOConnor, 2020-07-28:
It looks like you're using one of the experimental s-curve branches.  If you're having an issue with the BL-Touch, you'll need to reproduce the problem with the pristine code from this repo and submit a log with that.

Also, be aware of issue #2509.

-Kevin

1N4148, 2020-07-28:
Yes, saw that after posting. But with the original branch the problem still persists. Will supply the logs tomorrow.

samtayuk, 2020-08-04:
I'm having this issue too.

KevinOConnor, 2020-08-18:
I'm going to close this as there isn't much we can do without a log from the main Klipper branch.  To the best of my knowledge, the code works as intended.

-Kevin

closed 2020-08-18 by KevinOConnor