Klipper3d/klipper #3126: Erroneous and misleading documentation for screws_tilt_calculate command and [screws_tilt_adjust section]

BlazingBlackBeard, 2020-07-26:
The documentation for the screws_tilt_calculate command is specific in that it asks one to place the nozzle above the screw and record the absolute x y coordinates of the printer's position.

The existing documentation:
"To enable this feature, one would determine the nozzle coordinates near the screws and add them to the config file." should be corrected to:
"To enable this feature, one would determine the nozzle coordinates whilst the probe is directly above each screw (or as close as possible whilst the probe is still above the bed) and add them to the config file."

It also explains this wrong in example-extras.cfg:
...
#screw1: 100,100
#   The X,Y coordinate of the first bed leveling screw. This is a
#   position to command the nozzle to that is directly above the bed
#   screw (or as close as possible while still being above the bed).
...

This should be corrected to:
...
#screw1: 100,100
#   The X,Y coordinate of the first bed leveling screw. This is a
#   position to command the nozzle to that brings the probe directly
#    above the bed screw (or as close as possible while still being 
#    above the bed).
...

This current documentation and example is clearly wrong. At no time does it state that the probe should be above each screw, leading any sane person to conclude that the probe offsets provided in the [probe] or [bltouch] section of the printer.cfg are applied automatically by Klipper during the screws_tilt_adjust command. However, the probe offsets are not automatically applied during the screws_tilt_adjust command, resulting in probing off the bed, requiring one to manually edit and add the probe offsets to the previously determined nozzle absolute position x.y values in the [screws_tilt_adjust] section of the printer.cfg

Please edit the documentation to reflect that the probe should be above the screws, not the nozzle.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4978834/klippy.log)

KevinOConnor, 2020-07-28:
@rmcbc - any thoughts on this?

-Kevin

rmcbc, 2020-07-29:
@KevinOConnor, @BlazingBlackBeard is right because the coordinates is for the probe not for the nozzle.
When I can, I fix this. Sorry but I'm too busy and I'm a bit away from 3D printing. Give me some days and I fix it.

KevinOConnor, 2020-10-17:
FYI, I reworded the docs to try and make this more clear (commit abb0d880).

-Kevin

closed 2020-10-28 by KevinOConnor