Klipper3d/klipper #3127: bltouch 3.1 won't deploy pin on Z_TILT_ADJUST command

cata747, 2020-07-26:
Hi!
My printer was working until I flashed the last klippy firmware.
The homing part is working with bltouch but when reaches the Z_TILT_ADJUST it crashes into bed.
The nozzle is protecting the probe fortunately. 

I don't know what I am doing wrong. 
I hope someone can help me.
Thank you.

My printer: cartesian with skr 1.3 and TMC2209 sensorless homing on x, y

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4978893/klippy.zip)

KevinOConnor, 2020-07-28:
Make sure to run through the instructions found at https://www.klipper3d.org/BLTouch.html .  The most frequent cause of these types of issues are loose cabling or similar issues.

-Kevin

cata747, 2020-07-28:
Thank you for your response.
Everything was working. 
I will check the wirring.

closed 2020-07-28 by cata747