Klipper3d/klipper #3128: filament_sensor_switch section causes Klipper to crash on Replicape

goeland86, 2020-07-27:
I've added the following section to my config:
https://github.com/KevinOConnor/klipper/blob/25a9a32ecb03130ad6bc9282c8d126f9b1ae57a1/config/example-extras.cfg#L2027
Then uncommented the header and label sections to fill them in - resulting in the attached printer.cfg file.

However when I do so, Klipper starts, but then throws an error whenever it receives a G-Code or M-Code from OctoPrint, as can be seen in the attached klippy.log file:

```
Build file /home/debian/klipper/klippy/../out/klipper.dict(2878): Sun Jul 26 19:43:12 2020
Last MCU build version: v0.8.0-652-g25a9a32e-dirty-20200726_194312-kamikaze2
Last MCU build tools: gcc: (GCC) 10.0.1 20200315 (experimental) binutils: (GNU Binutils) 2.34
Last MCU build config: STEP_DELAY=2 MCU=pru CLOCK_FREQ=200000000 STATS_SUMSQ_BASE=256
Build file /home/debian/klipper/klippy/../out/klipper.elf(265028): Sun Jul 26 19:43:26 2020
Unknown command: buttons_ack
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
```

I have no explanation for why this is - the syntax looks correct, and I'm wired correctly on the X2 endstop with the Dyze Sentinel sensor (LED on the sentinel works correctly, suggesting it's reporting properly back to the endstop signal input).

[klippy.log](https://github.com/KevinOConnor/klipper/files/4983931/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4982236/printer.cfg.txt)

As soon as I comment out the relevant section and issue "restart" in the terminal, things go back to normal.

Arksine, 2020-07-28:
The PRU does not seem to support GPIO bit banging, thus its pins cannot be used as a "button".   The filament_switch_sensor uses button functionality to notify the host when the switch changes state.

I'm not extremely familiar with the BeagleBone/Replicape, does it expose any GPIOs via the linux "host" mcu?  If so you could connect to one of them, as it does support bit banging.

KevinOConnor, 2020-07-28:
That's correct - the PRU does not have enough space to compile in the "bit banging" code, but the host "linux mcu" can do these tasks.

-Kevin

goeland86, 2020-07-28:
@Arksine thank you for that summary.

I suppose I need to check if the linux MCU can map access to the endstop pins - which they should be able to. But I don't know if the PRU has exclusive access or not. Will need to test this a bit. But I could also just use an octoprint plugin for the filament sensor if I have to use GPIO access for it.

goeland86, 2020-08-12:
@KevinOConnor @Arksine I was able to just test & confirm that using the Linux MCU for the filament sensor does indeed work.
I'm putting my relevant config here (for a switch sensor wired to X2 output on the cape), in case someone else is interested in it.

```
[pause_resume]
recover_velocity: 50

[idle_timeout]
timeout: 1200

# Filament Switch Sensor.  Support for filament insert and runout detection
# using a switch sensor, such as an endstop switch.
[filament_switch_sensor sentinel_sensor]
pause_on_runout: True
#   When set to True, a PAUSE will execute immediately after a runout
#   is detected. Note that if pause_on_runout is False and the
#   runout_gcode is omitted then runout detection is disabled. Default
#   is True.
runout_gcode: outoffilament
#   A list of G-Code commands to execute after a filament runout is
#   detected. See docs/Command_Templates.md for G-Code format. If
#   pause_on_runout is set to True this G-Code will run after the
#   PAUSE is complete. The default is not to run any G-Code commands.
#insert_gcode:
#   A list of G-Code commands to execute after a filament insert is
#   detected. See docs/Command_Templates.md for G-Code format. The
#   default is not to run any G-Code commands, which disables insert
#   detection.
event_delay: 3.0
#   The minimum amount of time in seconds to delay between events.
#   Events triggered during this time period will be silently
#   ignored. The default is 3 seconds.
pause_delay: 0.5
#   The amount of time to delay, in seconds, between the pause command
#   dispatch and execution of the runout_gcode.  It may be useful to
#   increase this delay if Octoprint exhibits strange pause behavior.
#   Default is 0.5 seconds.
switch_pin: host:gpio30
#   The pin on which the switch is connected. This parameter must be
#   provided.

[gcode_macro outoffilament]
gcode: PARK_MACRO

[gcode_macro PARK_MACRO]
default_parameter_X: 0
default_parameter_Y: 0
default_parameter_Z: 50
gcode:
   G91
   G1 E-3.14 F1500
   G1 Z{Z}
   G90
   G1 X{X} Y{Y} F3000
```

goeland86, 2020-08-12:
For reference, [this particular chart](https://github.com/derekmolloy/exploringBB/blob/master/chp06/docs/BeagleboneBlackP9HeaderTable.pdf) and the [replicape pinout](https://wiki.thing-printer.com/index.php?title=File:Replicape_pinout_rev_B3.png), is what I think is what needs to be used to determine the proper GPIO numbers for the different endstops from the Linux MCU point of view.

closed 2020-08-12 by goeland86

KevinOConnor, 2020-08-13:
Might be worth adding the aliases to a `[board_pins]` config section to the config/generic-replicape.cfg example config file.

-Kevin