Klipper3d/klipper #3130: [Feature] CoreXYU, CoreXYUV, CoreXYUVAB , and Beyond Kinematics

eduncan911, 2020-07-28:
Thanks to @NordicGuy for bringing this up in #863, and thanks @KevinOConnor for the hint of hope! :)

After nearly two years since that was first asked, we now have Bill's (deckingman) CoreXYUVAB build and we are starting to see high-end CoreXYU and CoreXYUV builds with independent X carriages.  And now, Duet supports CoreXYUVAB within its firmware.

@KevinOConnor , could you point us in more of a direction about "porting new kinematics", which I'm assuming is replacing the rails classes?

We know CoreXY.  The "XY" is the kinematics around the x and y gantry.  RepRap defines CoreXYU has a 2nd gantry.  But we are seeing all sorts of builds now, and more letters of the alphabet being added.  To give a break down of some examples of what this alphabet soup means...

---

Now, imagine if you had a 2nd gantry mounted above the first.  In [Deckingman's build](https://www.youtube.com/watch?v=O_whcaAJpfs), his heavy 1kg hotend-only (of 6x inputs) operates on the standard CoreXY with precision.  He built a 2nd gantry ontop of the first that carries the 4kg weight of the extruders (so it's less precise).  That second gantry "mirrors" the first XY kinematics; however, it still needs to "home" independently of the XY gantry.  Therefore, the Duet guys created the UV kinematics to focus on homing the UV gantry (the extruders gantry).  

Then his massively tall machine started to rock back and forth printing at 300mm and 400mm speeds.  To address this, he built yet a 3rd taller level gantry - at the very top of the machine (6 feet tall now) that simply carries the conuter-weights that offsets the XY and UV gantries. This gantry continues to mirror the original XY kinematics; however, in reverse order this time (so to counteract opposing forces).  And, it also needs to home independently.  So the Duet guys created yet a 3rd set of XY kinematics, called the AB rails.

So this is where you get 3 different gantries, XY UV AB for the CoreXYUVAB setups.

---

If that wasn't enough, then you have other multiple-kinematics with insane setups such as the [BCN3D SIGMAX R19](https://www.bcn3d.com/bcn3d-sigmax-r19/) that uses normal Cartesian and/or CoreXY kinematics (not sure which).  However, they added a 2nd X mount/hotend that moves independently of the first X mount.  They call it something like the Independent Dual Extruder (IDEX) (I'm sure they have a trademark somewhere for that, lol).  

It's pretty wild though.  You can run it in "Copy" mode, that allows you to print two objects at the same time.  They have a "Mirror" mode that lets you print two objects at the same time like copy mode, but in a mirror of each other.  

And finally, you can have tool-changer support that allows you to switch between extruders much like normal dual hotends today.  However, this requires a 3rd "U" rail system, e.g. CoreXYU, to allow you to address those 2nd independent X movements now called a "U" rail.

**Where do we start with Klipper to support these, and other custom functionalities?**  :)  I personally have a Delta, and a CR-10 I can start to tinker with.  I also have a Dremel 3D20, that Klipper already has the ARM support of.  However, I need to figure out the bootloader and pinouts.

Fabrizio3, 2020-07-28:
Yup! developing a "CoreXYUV" here to use klipper and 2 skr boards, as I'll have 4 "XY" motors doing IDEX and independent 3 point Z leveling
my idea (taken from what has already been done on duet) was basically 2 coreXY systems running in one X rail, so the Y has to be synced but X is independent, also 2nd "Y" motor can't just mirror the first one as I'd like duplication as well as mirror and dual material modes.

[image]

pic for reference :)

eduncan911, 2020-07-28:
@Fabrizio3 I still think your description defines a CoreXYU - as you are only addressing a 2nd U rail, which is a 2nd X rail.

I still don't know how that would work quote yet with Klipper.  Is that a "Slicer" feature, to define gcode on X, Y, and U?  Do you submit two gcode files and is Klipper supposed to "merge/overlap" the Ys?  Etc.

I really think it's the former, in that the slicer us responsible for addressing the extra U axis in addition to the normal XYZ axis.  However, we need Klipper to support this new addressing system.  And, you need a slicer that supports CoreXYU!  Which, a quick Duck search doesn't show any hits on.

Deckingman's CoreXYUVAB is a much easier fish to fry to start with, as the UV and AB gantry addressing is only used in firmware, strictly for "homing" each axis.  Then when printing, they are simply setup to mirror XY.  

I'm just thinking here, as I'm not developing this feature (yet?  I may find time near the holidays).

I think starting with the independent homing of U, V, A, B in addition to X, Y and Z is a good start.  Because once those are defined, it would be an evolution to expose addressing each axis during printing.

Fabrizio3, 2020-07-28:
yeah tginkint about it my setup would be kinda weird, it's a XYU with 4 motors, as you have  XY and a YU systems, with independent pairs each

eduncan911, 2020-07-28:
Bad bot...

eduncan911, 2020-07-28:
> yeah thinking about it my setup would be kinda weird, it's a XYU with 4 motors, as you have XY and a YU systems, with independent pairs each

Oh, yours is not unique at all!  I had already ran across an OpenBuild of exactly what you are trying for.

https://openbuilds.com/builds/hlidskjalf-corexyu-dual-x-carriage.4378/

Fabrizio3, 2020-07-28:
Thanks! didn't catch this one, it's almost bang on! just that my budget is 1k for going 300mm/s+, it's the dream setup

KevinOConnor, 2020-07-28:
This looks like a dup of #1401 (and #2852 ).  I did my best to explain how to implement new kinematics in the developer documentation (see https://www.klipper3d.org/Code_Overview.html ).

-Kevin

eduncan911, 2020-07-28:
Thanks @KevinOConnor .  I guess search foo for CoreXYUV didn't hit the right keywords of CoreXYU.

I just finished reading all of them, plus the Code Overview page.  And me, like the rest, are not exactly sure where to get started implementing this.  

As you can see, there's a lot of interest in this ( total people now chiming in across all issues, and even more on /r/Klippers, and even more ok Discord if you search).  And, as I've explained in the OP, there are many ways this could be done.

So...  Could you give us some pointers?  "Start with X file, look at line 123, see how rail Y does..." kind of hints?

There are several of us ready to jump at this, we just your advice Obi-Wan to point us in the right direction!

n8bot, 2020-07-28:
For a good place to start, look at RRF's implementation of a generalized Core Kinematics: [CoreKinematics.cpp](https://github.com/Duet3D/RepRapFirmware/blob/dev/src/Movement/Kinematics/CoreKinematics.cpp)

KevinOConnor, 2020-08-01:
> So... Could you give us some pointers? "Start with X file, look at line 123, see how rail Y does..." kind of hints?

I don't know what needs to be implemented.  Figuring that out is a significant part of the work an interested developer would need to do.

-Kevin

rmie, 2020-08-05:
I'm wondering if this is working out-of-the-box, as it is very similar to multiple z-axis homing.

UV are the same direction/speed as XY
=> identical to z-axis homing, just ensure that motors/drives run the same length for each step

AB are opposite direction/speed as XY
=> not sure if it is possible to invert direction pins, but as a last resort reverse direction by flipping motor wires

One nice feature in this regard would be the option to share dir/step pins for multiple stepper drivers but have individual enable pins, to reduce the number of micro controller pins needed for cases like:
- multiple motor axis that require individual homing
- 2-in-1-out (no mixing) hotends, that require only one motor to run at any time

KevinOConnor, 2020-08-18:
I'm going to close this in favor of the existing #1401 .

-Kevin

closed 2020-08-18 by KevinOConnor

eduncan911, 2020-08-18:
Closing this issue may very well cause a 5th issue to be opened again, scattering the out even further.

May we recommend leaving this open and tagging it as a Needs Help?  So we have a centralized place to research and work on it, u til it's resolved with a PR?

Fabrizio3, 2020-09-14:
I've been taking a look at the documentation and the kinematics are written down as equations just like the image above, I'm wondering if it's as easy as just adding the U and defining the motor equations and relationships, or we'd need to do some development on accepting the 3rd axis and using a standardized definition so it works with cura or something, cause the equations are pretty simple and the Y "merging" is done directly in there, if it can read a "G1 U 35.6" for example and move the U I think we can hotwire cura to spit out idex code using U as a second axis, and make it do mirror, replication and dual material on the slicer side, I'm definitely looking into this soon-ish when I have some time to tinker, cuz the future is IDEX! lol

marcello-fialho, 2021-06-27:
Well at least implementation duplication and mirror mode for IDEX should be dead simple, as this can be done in hardware, by using logic gates. For duplication you just have to feed the step, dir and enable inputs to the X2 and E2 from the same source as the ones for X1 and E1. For mirror mode you do the same but invert the dir signal for X2. CoreXYUVAB is much more complicated, mirror and duplication mode are not special kinematics, they are just simple tricks. We have 5 different manufacturers of cheap IDEX printers in China now, actually 6 if we factor in the Biqu IDEX kit. Everybody wants input shaping, it is not in the radar for Marlin. So I guess next year all these manufacturers are going rrf.