Klipper3d/klipper #3131: Ender 3 firmware not flashed

hermannw, 2020-07-28:
Hi all,

I have here an Ender 3 with stock board currently running Marlin 2.0. The whole thing is controlled by a Raspi running Octoprint 1.4.0, Octopi 0.17.0 
I did the installation on the Raspi according to instructions and following the video "Teaching Tech" made.
Everything went ok up to the point where I tried to flash the firmware to the printer.
Here is the output:
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  Linking out/klipper.elf
  Creating hex file out/klipper.elf.hex
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 via avrdude

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: reading input file "out/klipper.elf.hex"
avrdude: writing flash (21924 bytes):

Writing | ################################################## | 100% 2.76s

avrdude: 21924 bytes of flash written
avrdude: verifying flash memory against out/klipper.elf.hex:
avrdude: load data flash data from input file out/klipper.elf.hex:
avrdude: input file out/klipper.elf.hex contains 21924 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.35s

avrdude: verifying ...
avrdude: 21924 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

So it seem like everything went ok - except it didn't. The LCD is "frozen" and after a new start it's just empty. I found several hints like restart Octoprint, reboot Raspi, reflash Marlin and try again. Nothing helped. I can reflash Marlin and everything works. I also tried to use avrdude to flash klipper.elf.hex. The result was the same: Flashing ok, verifying ok, no errors but nothing on the board.
Any ideas what went wrong?

Hermann

BlackStump, 2020-07-28:
[read carefully the top section of the ender3 config](https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg)
[you may need a boot loader](https://github.com/KevinOConnor/klipper/blob/master/docs/Bootloaders.md#atmega1284p)

hermannw, 2020-07-29:
Sorry I assumed that when I say I am using Marlin it's implied that I have a bootloader. Yes it's the Sanguino bootloader.

hermannw, 2020-07-29:
This was a misunderstanding on my side. I assumed that after flashing the firmware to the board, the display should show **something**. I realized later that the information in the display is coming from Klipper too. 
It works now. It shows again I should not **assume** but RTFM :-)
Thanks for sharing this great software!

closed 2020-07-29 by hermannw