Klipper3d/klipper #3133: Changing which .cfg file is used as the configuration file

Maverick1182, 2020-07-29:
Hi there,

I appear to have a semiworking config of Klipper working with OctoKlipper and my RaspberryPi.

The history is that I originally tried to install Klipper as per instructions but using my own user name "mark". Then, when it didn't appear to be working, I repeated all the steps thinking it may be to do with the permissions, so installed it under "pi".

I later realised the actual issue was the for some reason there was a permission issue with the /tmp directory preventing Klipper from writing the /tmp/printer file there. I appear to need to run "**sudo chmod -t /tmp**" to resolve this issue (please shout if there is a work around for this).

In any case, the **main** problem now is that the config file which appears to be used by Klipper is from /home/mark , however, if I point to it in OctoKlipper and try to make changes and save, they don't commit, perhaps because OctoPi is run as the "pi" user? This means I have to makes changes to the config file via WinSCP, and the ask OctoPi to restart for it to carry any changes correctly, which is a little clunky 

I was wondering whether there is a setting that tells Klipper which file to look for for the configuration and if I can point it to the /home/pi config so I can test whether it can be editted via the web interface? 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4995688/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4995692/printer.cfg.txt)

Arksine, 2020-07-29:
Klippy's arguments are set in `/etc/default/klipper`.  You can modify KLIPPY_ARGS to change where the config file is located.

Maverick1182, 2020-07-29:
Perfect, this worked when I sudo'ed into the file and replaced all references from the old user to "pi" and rebooted. Thanks!

closed 2020-07-29 by Maverick1182