Klipper3d/klipper #3134: z endstop always triggered after updating to the latest klipper

cyclonesurfer, 2020-07-29:
I made the required config changes regarding the servo that lifts the bed sensor.

I confirmed with QUERY_ENDSTOPS that it is always triggered, even when I unplugged the connector.

I also updated the SKR1.3 firmware built from the update.

klippy.log: https://pastebin.com/q6gGJgkN
[klippy.log](https://github.com/KevinOConnor/klipper/files/4997939/klippy.log)

cyclonesurfer, 2020-07-30:
This seems like a legit bug, it was working and then it wasn't after an update.

I queried the endstop in both states multiple times, all included in my klippy.log

The only thing I haven't done is a fresh install, but my understanding of git says that is not necessary.

BlackStump, 2020-07-31:
have you tried removing "^"  on your probe pin?
for your ref:
>Pin
 names may be preceded by an '!' to indicate that a reverse polarity
 should be used (eg, trigger on low instead of high). Input pins may
 be preceded by a '^' to indicate that a hardware pull-up resistor
 should be enabled for the pin. If the micro-controller supports
 pull-down resistors then an input pin may alternatively be preceded
by a '~'.

cyclonesurfer, 2020-07-31:
The pin has a hw pullup resistor installed.

I made that change and it was still unresponsive but showed as not triggered.

Sineos, 2020-07-31:
Could be a wiring problem (broken cable etc). If it is a NC switch it would report triggered on such a problem.

cyclonesurfer, 2020-07-31:
I verified continuity from the sensor pin to the board pin.

Nothing has changed except a software update and small config change to comply with the update.

cyclonesurfer, 2020-07-31:
Just tried a fresh install, did not fix, triggered no matter the condition of the signal on the pin.

KevinOConnor, 2020-07-31:
It's not clear which endstop you are having issues with.  However, there hasn't been any code changes to the endstop checking in years.  If you think there is a regression, try going back to the old code (for example, `cd ~/klipper ; git checkout af981432f ; make ; sudo service klipper stop ; make flash ; sudo service klipper start` ).  Then attach the log from the session with the old code as well as a session with the new code.  Please make sure the log session includes the QUERY_ENDSTOP and/or QUERY_PROBE commands.

-Kevin

cyclonesurfer, 2020-07-31:
I feel dumb, it is definitely a hardware issue, hooked up another switch and it registers it.

closed 2020-07-31 by cyclonesurfer