Klipper3d/klipper #3139: Help config filament sensor pin on Anycubic i3 Mega?

water-rhatt, 2020-07-30:
Can anyone tell me what to put under: 

`[filament_switch_sensor my_sensor]`->`switch_pin:`

I have tried both "ar5" and "ar19", based on what I could find from the Marlin firmware for this printer. In both cases a query returns 
```
Send: query_filament_sensor sensor=my_sensor
Recv: // Switch Sensor: filament not detected
Recv: ok
```

Also, I've noted that despite no filament detected the printer IS printing. Perhaps that is a clue as to what might be going on.

in Marlin config.h the section looks like this:
```
/**
 * Filament Runout Sensors
 * Mechanical or opto endstops are used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN for the first runout sensor.
 * For other boards you may need to define FIL_RUNOUT_PIN, FIL_RUNOUT2_PIN, etc.
 */
//#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW   // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  //#define FILAMENT_RUNOUT_DISTANCE_MM 25

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR
  #endif
#endif
```
and in the pins.h file:

```
/**
 * Arduino Mega with RAMPS v1.4 for Anycubic
 */

#ifndef BOARD_NAME
  #define BOARD_NAME "Anycubic RAMPS 1.4"
#endif

#define LARGE_FLASH        true

// Misc PINs
#define BUZZER 			   31
#define SDPOWER            -1
#define SDSS               53
#define LED_PIN            13
#define Z_MIN_PROBE_PIN     2
#define FIL_RUNOUT_PIN	   19
```
and, finally, my printer.cfg section is this:

```
[filament_switch_sensor my_sensor]
pause_on_runout: True
#   When set to True, a PAUSE will execute immediately after a runout
#   is detected. Note that if pause_on_runout is False and the
#   runout_gcode is omitted then runout detection is disabled. Default
#   is True.
runout_gcode:
 M117 Unloading Filament...
  G91
  G1 Z20 F2000
  G90
  G1
  M83
  G1 E10 F200
  G1 E-1.0 F1000
  G1 E1.5 F1000
  G1 E-1.5 F1000
  G1 E2.0 F1000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-100 F3000
  G1 E-80 F3000
  M82
  M400
  M117 Remove Filament Now!
#insert_gcode:
#   A list of G-Code commands to execute after a filament insert is
#   detected. See docs/Command_Templates.md for G-Code format. The
#   default is not to run any G-Code commands, which disables insert
#   detection.
#event_delay: 3.0
#   The minimum amount of time in seconds to delay between events.
#   Events triggered during this time period will be silently
#   ignored. The default is 3 seconds.
#pause_delay: 0.5
#   The amount of time to delay, in seconds, between the pause command
#   dispatch and execution of the runout_gcode.  It may be useful to
#   increase this delay if Octoprint exhibits strange pause behavior.
#   Default is 0.5 seconds.
switch_pin: ar19
#   The pin on which the switch is connected. This parameter must be
#   provided.
```

water-rhatt, 2020-07-31:
Not sure how this helps, but OK.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5005484/klippy.log)

Arksine, 2020-07-31:
It looks like the switch pin needs the internal pullup enabled.  Try adding to following to the [filament_switch_sensor] section:
```
switch_pin: ^ar19
```

FWIW, the filament_switch_sensor module will not prevent Klipper from printing when filament is not detected.  What it will do is execute the `runout_gcode` when the state transitions from "filament detected" to "filament not detected" during a print.

water-rhatt, 2020-07-31:
Thank you.  That plus reversing the pin worked: `switch_pi: ^!ar19`

closed 2020-07-31 by water-rhatt