Klipper3d/klipper #3140: When servo active the movement of the Z axis cause movement of servo (no jitter)

cicinovec, 2020-07-31:
Hi guys,

I have issue with servo that is switching my T0 and T1 nozzle.
When the servo is deactivated, it stays visually still, but I still hear constant repeating humming of it.
The worst case is when the servo is active - it actually moves even several milimiteres when moving of the Z axis.
I have observed biggest moves during homing of Z axis and during print it is with Z-hops moves

It is not electronic issue, I have checked it with scope and the servo responds to the angle command from MCU, that is changing within the software loop.

I have uploaded video to youtube.
The sequence visible there is homing of the Z axis when the servo is active. 
Watch the dark green bracket moving up and down
[https://youtu.be/y38B11n_XIs](url)

Klippy log is in attached zip, too big to be uploaded directly
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5007774/klippy.zip)

KevinOConnor, 2020-07-31:
It looks like you are "pushing the envelope" of an atmega2560 with respect to scheduling jitter.  Your dual-Z setup needs ~32K steps per second to move at 5mm/s, and the dual Z setup causes each Z step to schedule two step pulses at the same time.  If a Z step happens to coincide with a servo pwm update then some pwm jitter can occur; a pwm jitter of just 11us would result in a one degree swing of the servo.

Most people work around this by disabling the servo after moves (note that in the very latest code, disabling the servo is done via a `SET_SERVO ... WIDTH=0` command).  However, it seems your servo doesn't like the signal being disabled.

The only solution I can currently think of is to implement some kind of hardware change.  For example, switch to a different servo, use a different micro-controller, dedicate a low-cost micro-controller to the servo and use Klipper's multi-mcu support, etc.

-Kevin

cicinovec, 2020-08-01:
Hi Kevin,
Thank you for explanation, I try multiMCU solution now.

closed 2020-08-01 by cicinovec