Klipper3d/klipper #3142: MCU 'mcu' shutdown: No next step after a few minutes of printing

lkong, 2020-08-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5009755/klippy.log)

Symptoms:
Stop printing after a few minutes into printing.

Host version:
commit 25a9a32ecb03130ad6bc9282c8d126f9b1ae57a1
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Sun Jul 26 10:41:43 2020 -0400

    docs: Move SET_TEMPERATURE_FAN_TARGET to its own section in G-Codes.md

    Signed-off-by: Kevin O'Connor <kevin@koconnor.net>

Binary version is up to date.

lkong, 2020-08-01:
I think input shaper is the root cause here. I had the config as 
shaper_freq_x: 67.4
shaper_freq_x: 71.6

Once the config is removed, the printer prints fine.

KevinOConnor, 2020-08-04:
FYI, this does look like a Klipper bug.  It will likely take a few days for me to track it down though.

-Kevin

dmbutyugin, 2020-08-06:
@lkong Could you also post the gcode you were trying to print?

KevinOConnor, 2020-08-06:
FYI, a "no next step" error occurs when the micro-controller detects that a stepper motor goes to zero velocity directly from a speed greater than a configured limit.  (It's intended to catch communication outages that could cause a stepper motor to empty its motion queue while still running at speed.)  However, I fear the input_shaper code may have changed the kinematics so that the current maximum deceleration checks are no longer valid.  I still need to look through the math and confirm though.

-Kevin

slavedas, 2020-08-06:
Just FYI I have the same issue and the same solution.  Comment out the Input shapers and it goes away. Happens to me when printing relatively large amounts of rectilinear bottom infill.  The ringing tower printed fine, but a drawer bottom did not.  Error replicates with 100% reliability until input shaping is turned off.  I'm on a CoreXY printing at 100mm/s first layer if that helps at all.  I can try turning down the first layer.  Also I had a second basically unused MCU connected and it got disconnected and threw the error first, until I removed it from my config and THEN the main MCU reported the disconnection.

dmbutyugin, 2020-08-06:
@KevinOConnor FWIW, input shaping cannot exceed the configured limits (for positive-valued shapers, which we use here): instantaneous velocity and acceleration can never be larger than the maximum unshaped values. This is because the shaped position (and velocity and acceleration accordingly) is \sum c_i * x(t + t_i), where all c_i >= 0 and \sum c_i = 1. I'm not 100% certain about the instantaneous velocity jumps, but I think the same logic should apply. But it would greatly help with debugging, if you @lkong or @slavedas could share the problematic gcode you've been printing (and klippy.log, @slavedas).

dmbutyugin, 2020-08-06:
So, my point is: it is also possible that there are some other bugs in input shaping that manifest this way (e.g. when determining kinematic activity), and it is also possible that input shaping triggers some other bug elsewhere.

robthide37, 2020-08-06:
could raising accel be a temp fix ? I run high accel and haven't ran into this .

slavedas, 2020-08-06:
Fair requests @dmbutyugin they are both included.  Notice my print is literally just a single flat layer.  Dies very quickly on the first corner.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5037088/klippy.log)

[Shape-Box.zip](https://github.com/KevinOConnor/klipper/files/5037090/Shape-Box.zip)

n8bot, 2020-08-06:
I notice that you are using a slic3r-derivitve, and you say that the problems are in rectilinear solid infill.

I'm currently diagnosing a problem with PrusaSlicer rectinliear infill, where certain moves will suddenly change extrusion rate, in the middle of an infill pattern (usually near the edge).

I'm not sure if this could at all be affecting anything with the issue, but it's something to be aware of. Most slicers do not produce consistent extrusion rate at all. There could be some superfluous acceleration caused by this.

slavedas, 2020-08-06:
> I notice that you are using a slic3r-derivitve, and you say that the problems are in rectilinear solid infill.
>
> I'm currently diagnosing a problem with PrusaSlicer rectinliear infill, where certain [...]

I'm using SuperSlicer with Filled-rectilinear infill, I could definitely turn off the filling and see if that makes a difference.

n8bot, 2020-08-06:
I bet if you analyze your GCode by manually calculating the extrusion rate based off the vector lengths of the moves, you'd find lots of sudden increases and decreases in your extrusion rate. I've worked to reduce this in custom builds, but some parts of infill are still probelmatic.

lkong, 2020-08-06:
> 
> 
> @lkong Could you also post the gcode you were trying to print?

[V350_ringing_tower.log](https://github.com/KevinOConnor/klipper/files/5037480/V350_ringing_tower.log)
Attached is the gcode, rename it to .gcode to return it to its original form.
The slicer I use is Cura 4.6.1

lkong, 2020-08-06:
> Come to think of it, with stock SuperSlicer, the variation of extrusion rate should be present in all feature types, so it wouldn't be noticeably worse in infill, so this is likely not the issue. (Or maybe it is noticeably worse in infill? I havent analyzed stock slic3rs in a while I've been comforably playing with 6 and 9 digits of decimal precision. lol)

The shutdown is deterministic and definitely didn't happen during infill. As matter of factor, gcode provided is sliced as vase mode.

dmbutyugin, 2020-08-06:
@robthide37 
> could raising accel be a temp fix ? I run high accel and haven't ran into this .

No, unfortunately, I do not think so. Changing the parameters may surely mask the problem in a particular case, but I think the bug is triggered by some combination of printer parameters and the specific sequences of gcode commands.

Thanks all, I think it should be reasonably easy to reproduce the problem now.

dmbutyugin, 2020-08-06:
@KevinOConnor I think it gets pretty interesting. I ran the gcode posted by @slavedas simply at 25a9a32ecb03130ad6bc9282c8d126f9b1ae57a1 (with some debugging logging added as bab9dea42a70ea5c694c60cda5cc1a3dc60504cd), with the config:
[printer.log](https://github.com/KevinOConnor/klipper/files/5038443/printer.log)
and got some pretty interesting results. I think the iterative solver sometimes, well, does not really hang (as it eventually progresses further), but runs into some numeric (stability?) issues. The extra logging that I get is (an excerpt):
```
INFO:root:flushing queue (4 moves)..
INFO:root:process moves (2 moves, flush took 0.000015553 sec)..
INFO:root:queue flushed (process moves took 0.001500813 sec)
INFO:root:flushing queue (4 moves)..
INFO:root:process moves (2 moves, flush took 0.000016887 sec)..
INFO:root:queue flushed (process moves took 1.646276180 sec)
INFO:root:flushing queue (4 moves)..
INFO:root:process moves (2 moves, flush took 0.000020874 sec)..
INFO:root:queue flushed (process moves took 0.002042876 sec)
```
So, normally flushing takes tiny fraction of second. However, occasionally the call to `self.toolhead._process_moves(queue[:flush_count])` takes a lot of time (~1-2 seconds on i7-7500U, so not too weak). And it's not like there's lots of moves to process. I think the same happens on RaspberryPi as well, just there the flush takes considerably longer, and the mcu simply really runs out of scheduled steps.

KevinOConnor, 2020-08-07:
@dmbutyugin - Thanks.  I was able to reproduce locally and I've found the root cause.  Unfortunately, for some reason with the given configuration, the "false position method" is turning into a linear search.  That search is taking an extremely long time to complete.  The problem is described at https://en.wikipedia.org/wiki/Regula_falsi#Improvements_in_regula_falsi .  I didn't think this would be a problem in practice in Klipper, but it looks like we've hit that corner case.

I'll need to think about the best way to handle.  I'm also open to ideas.

-Kevin

dmbutyugin, 2020-08-07:
@KevinOConnor Hmm, well, yes, maybe it's just an unfortunate set of conditions. It is quite possible that they could have been achieved even without input shaper, but we just never ran into such cases before.

Well, that page claims that "Regula falsi's failure mode is easy to detect: The same end-point is retained twice in a row". So maybe we can track that condition throughout the search, and if it is triggered, then either switch to some improved version, or (I think even better) to a bisection algorithm. I think after a few iterations (like, 5-10), if the precision was not already reached, we can probably try to switch back to the "false position method", hoping that those unfavorable conditions have been overcome. And we can use the existing tests to check if that assumption (that some few iterations of a less efficient bisection method are sufficient to return the false position method to a normal mode of operation) is correct or not.

KevinOConnor, 2020-08-08:
We could adopt the "Illinois algorithm".

```diff
--- a/klippy/chelper/itersolve.c
+++ b/klippy/chelper/itersolve.c
@@ -22,7 +22,7 @@ struct timepos {
     double time, position;
 };
 
-// Find step using "false position" method
+// Find step using "false position" method (with "Illinois algorithm")
 static struct timepos
 itersolve_find_step(struct stepper_kinematics *sk, struct move *m
                     , struct timepos low, struct timepos high
@@ -39,6 +39,7 @@ itersolve_find_step(struct stepper_kinematics *sk, struct move *m
     if (high_sign == signbit(low.position))
         // The target is not in the low/high range - return low range
         return (struct timepos){ low.time, target };
+    int prev_choice = 0;
     for (;;) {
         double guess_time = ((low.time*high.position - high.time*low.position)
                              / (high.position - low.position));
@@ -51,9 +52,15 @@ itersolve_find_step(struct stepper_kinematics *sk, struct move *m
         if (guess_sign == high_sign) {
             high.time = guess_time;
             high.position = guess_position;
+            if (prev_choice > 0)
+                low.position *= .5;
+            prev_choice = 1;
         } else {
             low.time = guess_time;
             low.position = guess_position;
+            if (prev_choice < 0)
+                high.position *= .5;
+            prev_choice = -1;
         }
     }
     return best_guess;
```

When I first implemented the iterative solver a couple of years ago, I had some reservations about the "Illinois algorithm".  I don't immediately see why it would not work well now, however.  It does fix the issue here.

-Kevin

KevinOConnor, 2020-08-08:
> FWIW, input shaper maintains that all moves are 2-nd order polynomials, it just does the linear interpolation with some offset as \sum c_i * P(t + t_i)

FYI, the problem occurs because input_shaper impacts the step generation of the previous move.  So, if there is a Y=0mm/s move followed by a Y=20mm/s, then when the iterative solver calculates the steps for the Y=0mm/s move, it comes across a non-smooth transition a few milliseconds prior to the end of the Y=0mm/s move.

I believe the same problem could occur with smoothPA, but I don't think it could occur prior to that.

-Kevin

dmbutyugin, 2020-08-08:
@KevinOConnor 
> When I first implemented the iterative solver a couple of years ago, I had some reservations about the "Illinois algorithm". I don't immediately see why it would not work well now, however. It does fix the issue here.

I am, unfortunately, not very familiar with these algorithms to suggest a qualified judgement - in particular, how it stands with respect to normal false position and bisection methods. But if it works better than bisection method and guaranteed to not have failure modes, then it would make sense to use (though Wikipedia page claims that other than bisection, all of the numerical equation-solving methods can have a slow-convergence or no-convergence problem under some conditions).

> FYI, the problem occurs because input_shaper impacts the step generation of the previous move. So, if there is a Y=0mm/s move followed by a Y=20mm/s, then when the iterative solver calculates the [...]

It does affect the generation of the previous move indeed. However, one could equivalently split a move near the beginning and the end further into a few moves with piecewise smooth quadratic X/Y coordinates (similar to the current split of the trapezoid into 3 parts - acceleration, cruising, and deceleration). Also, there are velocity discontinuities only if the original boundary between the moves has a velocity jump on one of the coordinates (e.g. around the corner). So I am not sure if input_shaper changes anything in this regard. But it does add more velocity jumps, and with different (smaller) jump magnitudes - so it maybe has more chances of triggering unfavorable conditions?

KevinOConnor, 2020-08-08:
FYI, I committed the change to use the updated "Illinois algorithm" (commit 513582af) along with stricter completion checks (commit ea5e7674).

I think this should fix the problem reported here.

-Kevin

KevinOConnor, 2020-08-08:
> But if it works better than bisection method and guaranteed to not have failure modes, then it would make sense to use (though Wikipedia page claims that other than bisection, all of the numerical equation-solving methods can have a slow-convergence or no-convergence problem under some conditions).

I think it should work okay.  As I understand it, the failure case is where the stepper isn't moving at all for most of the search interval, but then has a non-smooth transition resulting in movement at the very end of that interval.  This would cause the "false position method" slope calculation to always report a far away guess.  The "Illinois algorithm" will rapidly converge in this situation.  I'd guess the wikipedia article is referring to possible "weird" math functions, but all of our functions are sane.

> Also, there are velocity discontinuities only if the original boundary between the moves has a velocity jump on one of the coordinates (e.g. around the corner).

Alas, the way I coded the iterative solver, it was also susceptible to acceleration discontinuities within a linear move.  The updated boundary checks (commit ea5e7674) should narrow that window.  (There's still the possibility of a "forward than backwards motion" being skipped by itersolve_gen_steps_range(), but I don't think that will be a problem in practice.)

-Kevin

dmbutyugin, 2020-08-09:
> Alas, the way I coded the iterative solver, it was also susceptible to acceleration discontinuities within a linear move. The updated boundary checks (commit [ea5e767](https://github.com/KevinOConnor/klipper/commit/ea5e76746a8f0ea72d014b715ac4712bd47513de)) should narrow that window. (There's still the possibility of a "forward than backwards motion" being skipped by itersolve_gen_steps_range(), but I don't think that will be a problem in practice.)

I see, I did not realize it. This was the reason to split a move into acceleration, cruise, and deceleration part then, I guess. Then indeed, it is possible that the input shaper would cause this type of issue. SmoothPA could also be susceptible, though potentially to a lesser degree due to the proper choice of the smooth function (vanishing at the ends of a smoothing interval).

KevinOConnor, 2020-08-10:
> I see, I did not realize it. This was the reason to split a move into acceleration, cruise, and deceleration part then, I guess.

Well, to be fair, I only realized it yesterday..  The `struct move` split was for code convenience.  In practice I doubt that change makes a difference, because I think the issue would only occur in practice when going from zero movement to some movement mid linear move.  

And because on most kinematics the interative solver skips calculations when a stepper isn't moving at all, I suspect it really would only show up on corexy.  On corexy a diagonal move, for example, may not move the "a+b stepper", but the iterative solver will still get called - that opens up the window for the "no movement" to "some movement" within a single linear move.

In any case, hopefully this will no longer be a problem with the latest changes.

Cheers,
-Kevin

KevinOConnor, 2020-08-24:
I'm going to close this as we think the Klipper bug is now fixed.

-Kevin

closed 2020-08-24 by KevinOConnor