Klipper3d/klipper #3151: [Feature] Odrive support. Open source Budget closed loop servo drive.

bung69, 2020-08-03:
[https://odriverobotics.com/](url)
[https://www.youtube.com/watch?v=WT4E5nb3KtY&feature=emb_logo.](url)

Hi all, 

I am considering purchasing a couple of Odrives for use on a larger machine that will be controlled by Klipper.

An Odrive can be controlled by Step and direction,  though it seems a bit backwards and wasteful to use another MCU in front of the odrive to just generate step and direction. There would also be some additional latency on the servo's vs any other Steppers.

Portions of klipper's stm32 firmware could lightly be ported / merged with the Odrive firmware (also stm32) to allow an Odrive to work with klipper just like an extra MCU,  with motor temperature feedback and potentially velocity, torque, current and position feedback. Torque on an extruder motor could potentially be used to detect jams. 

I am curious if there is any interest in this and if anyone could give me pointers on where to start or tell me its not worth it  / too much work / i am barking up the wrong tree etc.  I dabble with C  and Python on occasion,  but mostly just hacking together bits of existing code.   

To my novice eyes it looks like at a minimum  the timer, scheduler, serial, and probably stepper modules would need to be ported across. 

Cheers.

KevinOConnor, 2020-08-04:
I don't know of anyone that's looked at running Klipper on the odrive.  However, there was some initial work done on the mechaduino (and similar) servo stepper motors.  For example, see: https://github.com/KevinOConnor/klipper/pull/1038 .  This was also recently discussed in #3103 .

-Kevin

bung69, 2020-08-14:
Hi Kevin,  Thanks for the links.   i've had a look through and am starting to get my head around the source.  

In virtual_stepper.c  you comment that it would be possible to modify the step compression code,   i'm guessing you would just send the target position from the host rather than track steps on the mcu?

KevinOConnor, 2020-08-14:
> In virtual_stepper.c you comment that it would be possible to modify the step compression code, i'm guessing you would just send the target position from the host rather than track steps on the mcu?

Yes.  For a traditional stepper motor driver, the micro-controller ultimately needs to know how many clock ticks there should be between step pulses.  In contrast, the mechaduino code (as currently implemented) ultimately needs to know what position the stepper motor should be at on each control loop iteration (a 6khz loop).  The current mechaduino code uses timers to convert the "time between step" into "position at each loop iteration".  However, it should be possible for the host to calculate the position the stepper should be at on each loop iteration, compress that, and then send that schedule to the micro-controller.  That would simplify the micro-controller code.

-Kevin

closed 2020-09-13 by github-actions[bot]

theolas, 2021-05-17:
More and more people are starting to use odrive together with Klipper. I think this should be revisited and implemented.

eric-schleicher, 2021-05-17:
Note:   ^ i liked this thread before it was cool.     @theolas is correct, there is quite a bit of interest in this!

arturohernandez10, 2021-05-28:
I visited the odrive site and the comments from the odrive experts are that step/dir is the lowest level denominator. I think this is because they are looking at a low level pin connection. But Kevin. You send velocity and acceleration to the controller.

queue_step oid=7 interval=7458 count=10 add=331

If I recall correctly these commands are meant to compress the info and give precise timings. But they can be converted to distance, velocity and acceleration. The only other comment from the odrive group. Is that they would prefer to receive the commands at regular intervals. I'm not certain that this is a huge issue since the accel odds included. But it would be simple to take those commands buffer them and emit equivalent commands at regular time intervals if needed. It's only a matter of forwarding the command to odrive from the firmware. I just have not seen this mentioned here. I may be to late tip the party. Cheers.

arturohernandez10, 2021-05-28:
Just in case I'd be happy to write the buffering/translation code. I imagine 1 or 2 seconds delay would be more than enough. It could be written in python as a forwarding command to the firmware. The delay would need to be applied to all the signals. If a lot of people use the feature. Someone could do the work to produce the new signal without the delay.

KevinOConnor, 2021-05-28:
Yes - the host code always knows the desired position of the motor and has access to upcoming positions.  Thus it can calculate position, velocity, and acceleration.

How that information gets to the motor is up to the implementation.  One could port the Klipper micro-controller code to the odrive hardware, or one could extend the Klipper code to send messages to the odrive firmware.  Again, those details would be up to the persons implementing that support.

FWIW, I suspect few people will see this ticket on github - if anyone is interested in discussing it, a topic on the Klipper Discourse may get more visibility.

-Kevin

wuast94, 2021-09-18:
is there anything new? i have a huge printer and had a badly failure last night, an closed loop stepper would prevent this, and this odrives seems to be the best option price/performance wise, i would start to implement it myself but i cant code 😅

bung69, 2021-09-19:
Vez3D youtube channel has some videos about using odrive with klipper on a
3d printer.  Up to 1m/s @ 50k accelleration, however he also outlines some
of the problems with odrive when used on a 3dprinter, posibly related to
FOC or cogging.

As much as it is a nice idea to intergrate klipper firmware directly on to
the odrive board in some way,  it dosnt seem necessary as odrives the step
/ direction input seems to work fine.

Shouldnt be that hard to make a python add on to get some of the odrive
stats, error info etc back in to klipper / webui though, even for a non
programmer.