Klipper3d/klipper #3152: Can't connect to MKS RUMBA32 via USB after configuration

Bernouillle, 2020-08-04:
Hello everyone,

i switched for printer setup from a RUMBA+ board to an MKS RUMBA32. After i managed to flash the controller board i wasn't able to find the board via USB connection anymore. I used the same configuartion as in #2298.
Also, i didn't change the configuartion via USB port in the "make menuconfig" setup. Does someone have a clue how i can comminucate with my board again?
Thanks a lot in advance, guys!

P.S.: Unfortunately, i am not able to provide you with a screenshot atm. Would take some time as i'm not in the office currently.

Arksine, 2020-08-05:
I don't have this device, but my guess is that you can enter the DFU Bootloader by holding the boot button during a reset.

As the issue you linked indicated, it appears that this board uses a 12 MHz crystal.   If you had 8 MHz selected it would explain why USB is not enumerating.

Bernouillle, 2020-08-05:
@Arksine yeah, that´s how you can enter the Bootloader. But it also has a USER button, which is not on the original board from Aus3D and i am not sure what it really does. I took the issue i posted as example to setup my RUMBA32. But i am not sure, if my board uses a 12 MHz crystal too, because it is from another vendor. Do you know how to check about the used crystal rate in the documents of the board? I am really knew to this stuff. Here is the GitHub page of the mks RUMBA32: https://github.com/makerbase-mks/MKS-RUMBA32

KevinOConnor, 2020-08-05:
The only way to determine the crystal frequency is via the schematic.  The schematic from that site indicates the board has a 12Mhz crystal.

-Kevin

closed 2020-08-11 by klipper-gitissuebot