Klipper3d/klipper #3154: cycle_time suggestion

ehtnevets, 2020-08-04:
Can someone add an extra note on cycle_time documentation for heaters (especially heater_bed) to decrease (thus increase frequency) if flickering of house lighting is observed. 10Hz default for me causes very noticeable flickering on my LED ceiling lights.

ehtnevets, 2020-08-05:
At 125Hz (cycle_time: 0.008), I don't observe any flickering on my room LEDs.

robthide37, 2020-08-07:
is that to fast for ssr ?

ehtnevets, 2020-08-07:
Mine uses mosfet. I think most printers use mosfet.

robthide37, 2020-08-07:
i am powering my printer and house with a generator roght now because my town and alot of the state has no electric for amost week now . Also my bed is wired to ac mains and uses a ssr . my hotend is using the boards mosfet output. Technically i am using the mosfet and the ssr with mains voltage for the bed and yes right now i can get flicker with the generator but if i raise pwm or drop to .3 it isn't that bad but at high pwm like .01 or faster i get no flicker i just am worried about the ssr

ehtnevets, 2020-08-08:
Probably 10ms is your max. If you know the type of SSR you can check its datasheet for t(on) time. You can always try something like this, https://www.amazon.com/gp/product/B07TWL385B if you want it faster pwm. 

I'm more sensitive to flickering than most people so 125Hz is my minimum.

nikkoaws, 2020-08-10:
> Probably 10ms is your max. If you know the type of SSR you can check its datasheet for t(on) time. You can always try something like this, https://www.amazon.com/gp/product/B07TWL385B if you want it faster pwm.
> 
> I'm more sensitive to flickering than most people so 125Hz is my minimum.

Most SSR are zero crossing meaning they will stay on until the AC input goes to zero volts - so switching faster doesn't really work unless you sync it with the mains 50/60 Hz frequency

ehtnevets, 2020-08-10:
That's AC SSR. The DC SSR is not.

closed 2020-08-11 by klipper-gitissuebot