Klipper3d/klipper #3156: [Question] Possible to somehow "link" input_max to max_temp?

th33xitus, 2020-08-05:
I noticed that the default Klipper temperature menu has a harcoded `input_max` for the temperature.
For the extruder you can set a `max_temp` value.

I was wondering if one can somehow "link" those two values together. Let's say i am switching my hotend from one with a heatbreak with a PTFE insert to a fullmetal one. The first one can take 250°C max. and therefore the `input_max` should be that exact value, or for safety reasons 10°C lower than the `max_temp`. The fullmetal one on the other hand can easily take 280°C or more, therefore the `input_max` can be higher than 250°C.

### So my question is:
Is there a way, that the `input_max` value is altered when the `max_temp` value is changed? 
Like, let's say, `input_max = (max_temp - 10)`

[image]

I tried to fiddle around already and set `input_max: {printer.configfile.config["extruder"]["max_temp"]}`, but that did not work at all.

Is there another way to achieve that? Or is impossible, no matter what?

KevinOConnor, 2020-08-05:
This is discussed a bit in #1856 and #2344.  We plan to merge #2344 in the next few days.

-Kevin

KevinOConnor, 2020-08-18:
FYI, this should be fixed with the merge of #2344 .

-Kevin

closed 2020-09-17 by github-actions[bot]