Klipper3d/klipper #3160: Connecting multiple nodes without serial connection not documented

cHolzberger, 2020-08-06:
i just burned my last arduino uno by accident but have some bananapi's, rpis and other stuff with gpio laying around. i'm trying to use those other boards as "thick-mcu's" using their gpio pins.

but i can't find any straight documentation on this use-case. it seems some code exist to handle those cases e.g. https://github.com/KevinOConnor/klipper/blob/db05938a5fd04206ec297c816f5f7fbd0a86e421/klippy/mcu.py#L403 

/dev/rpmsg_ seems to be OpenAMP framework
/tmp/klipper_host_ seems to be some sort of remote socket connection 

it would help if someone who knows how to handle those things could write some documentation about it.

jakep82, 2020-08-06:
I'm not sure what you're trying to accomplish, but documentation for using a Pi as a secondary MCU does already exist. 

https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md

cHolzberger, 2020-08-07:
yeah this is the pi that klipper is running on? i thought a second pi with its gpio pins can be used instead of a microcontroller. 

i want to do a setup like: 

PC
|- USB -> AVR  -> (X Servo, X Endstop , ... , Temp Sensor ) 
|- Network -> RPi GPIO Pins -> (Z-Servo, Z-Endstop, Z-Probe)

i somehow expected this to work somehow but the more i read the documentation this isn't a planned scenario

jakep82, 2020-08-07:
The Pi is only meant to be used as a secondary MCU for controlling non-realtime functions like fans and lights.  You still need a microcontroller to handle things like steppers and endstops.

KevinOConnor, 2020-08-07:
I don't know of anyone that has attempted to use the "linux mcu" code on a separate machine from the primary host machine.

It's technically possible to use tools like socat to redirect the /tmp/klipper_host_mcu connection over tcp.  As indicated above, be aware that the "linux mcu" code isn't intended for high throughput tasks.

Cheers,
-Kevin

closed 2020-09-05 by github-actions[bot]