Klipper3d/klipper #3161: MCU Shutdown during printing, with error: MCU 'mcu' spontaneous restart

dwisetiyadi, 2020-08-06:
Hi,

I have a problem, MCU always going shutdown when printing. I've change connection from USB to serial RX/TX & Ground, but the problem still same.

Attached my [klippy.log](https://github.com/KevinOConnor/klipper/files/5036946/klippy.log)

KevinOConnor, 2020-08-07:
The most common reason for this is wiring problems - often the head movement can pull on a loose wire or press against a surface which causes a wiring short.  It's also possible there was an incomplete micro-controller flash - you could try building the firmware again and reflashing it (`make clean ; make flash`).

-Kevin

dwisetiyadi, 2020-08-07:
> The most common reason for this is wiring problems - often the head movement can pull on a loose wire or press against a surface which causes a wiring short. It's also possible there was an [...]

Thanks for your response. Wires looks good, but i'll check it again. I tried 3 times to clean reflash the mcu firmware, but the same problem still comes. Should I change the mcu with new one?

BeatSlayer, 2020-08-07:
Maybe a voltage drop, I would check them with a multimeter.

dwisetiyadi, 2020-08-08:
> Maybe a voltage drop, I would check them with a multimeter.
> 
> EDIT:
> You could also try a dry run, in case of it being a voltage drop it will be less likely for it to happen if not using heaters.

Hi,

I measure the input voltage using multimeter during printing, its 5.5v on Raspberry, and 23,5v on MCU (I used SKR 1.3), is it normal?

dwisetiyadi, 2020-08-08:
Problem solved, just like @KevinOConnor said, its wiring problem. I'm gonna to close this, many thanks.

closed 2020-08-08 by dwisetiyadi