Klipper3d/klipper #3163: Support for MKS Robin Nano v2.0 + MKS TS35 v2.0 (Touch LCD)

msource, 2020-08-07:
Hello,

I don't know if this is the correct place to ask this. How can I help to have the new MKS Robin Nano v2.0 + MKS TS35 V2.0 Touch Screen supported by Klipper?

Sincerely,
Marcos Fernandes

robthide37, 2020-08-07:
it is

KevinOConnor, 2020-08-11:
There is support for some MKS Robin nano v1.2 board, but I don't know if the v2 is supported and I don't know if the display is supported.

https://github.com/KevinOConnor/klipper/blob/master/config/generic-mks-robin-nano.cfg

-Kevin

robthide37, 2020-08-11:
the display now uses the aux cables so can emulate reg graphic display and
the board itself is just changes to the display connector so now will work
with btt and reprap displays and it added breakout connector for cls and
traces for uart without flying cables but otherwise its the same as the 1.2

closed 2020-09-09 by github-actions[bot]

anteyru, 2021-09-18:
> the display now uses the aux cables so can emulate reg graphic display and the board itself is just changes to the display connector so now will work with btt and reprap displays and it added [...]

Not so simple. The TS35 display is still not supported.

Spede90, 2022-01-22:
I just bought MKS Eagle(based to the MKS Robin Nano V3) with TS35 included. Any updates regarding the TS35 support in near future?