Klipper3d/klipper #3166: Bed not sttopping when homing Z with piezo

definitely-not-a-t-rex, 2020-08-08:
I'm trying to get my precision piezo to work with klipper, I modified the z endstop to be the probe, set up probe and mesh but when I try homing Z the bed doesn't stop and I have to either turn the printer off or issue an M112 to stop it. I'm using and skr 1.4 turbo with tmc 2209 drivers on my modded ender 5 pro. X and Y work fine (I still have to calibrate the value for sensorless homing but they move and home). The pcb of the piezo is connected 5v to z stop's 5v, gnd to gnd and sig to z stop's sig (P1.27 according to btt's github) and when I touch the nozzle the led turns off and then on again as it should. It's the last part that I need fìto fix to finally get my printer working on klipper 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5046080/klippy.log)

jdlongenecker, 2020-08-08:
Hello! have you checked QUERY_ENDSTOPS to make sure that Klipper is reading when the piezo sensor is triggered?

definitely-not-a-t-rex, 2020-08-08:
> Hello! have you checked QUERY_ENDSTOPS to make sure that Klipper is reading when the piezo sensor is triggered?

How can I check if it sees the trigger? If I use the command while not probing it shows x/y/z as open  and if I try touching it it doesn't change but I haven't tried it while probing because it hard crashes into the bed and I don't want to destroy the hotend/bed since it make an awful noise

tobyfu, 2020-08-09:
I'm not certain this will be the solution but I have a similar setup (different printer but skr1.4 turbo with 2209s and precision piezo as Z probe). There was something about needing to remove the DIAG pin from the 2209 drivers if you were not going to use sensorless homing on that particular axis otherwise the endstop pins would be ignored. I didn't want to modify my drivers so instead of using the Z endstop pin for the probe I used the pins labeled PWRDET on the pinout  (P1.0) and that worked for me.

This is my [probe] section:

[probe]
pin: ^!P1.0
x_offset: 0.0
y_offset: 0.0
speed: 4.0
samples: 2
sample_retract_dist: 5.0
lift_speed: 7.0
activate_gcode: G4 P500

BlackStump, 2020-08-09:
Why not use the dedicated [probe port P0.10](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/SKR-V1.4-Turbo-pinout.jpg)

definitely-not-a-t-rex, 2020-08-09:
> Why not use the dedicated [probe port P0.10](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/SKR-V1.4-Turbo-pinout.jpg)

by itself it didn't fix the issue but with @tobyfu solution either P1.0 or P0.10 works

definitely-not-a-t-rex, 2020-08-09:
> I'm not certain this will be the solution but I have a similar setup (different printer but skr1.4 turbo with 2209s and precision piezo as Z probe). There was something about needing to remove the [...]

I had to also lower VR2 so that it was just before turning red and not after but this with ^! on the pin solved the problem, thanks!

closed 2020-08-09 by definitely-not-a-t-rex