Klipper3d/klipper #3167: New Cancel_print in Pause_resume.py conflict.

AbomShep, 2020-08-09:
I've used a cancel_print macro since the day I first used Klipper. DWC2 recognizes this macro and has a dedicated button on it's screen for it. I recently tried to enable the filament runout sections and discovered that cancel_print was recently added as a command in the pause_resume.py file, and this conflict crashes Klipper.  I suggest that the internal name used be changed for this reason.

[capture.log](https://github.com/KevinOConnor/klipper/files/5048163/capture.log)

Arksine, 2020-08-09:
You can override pause_resume's CANCEL_PRINT with gcode_macro's `rename_existing` option.  For example:
```
[gcode_macro CANCEL_PRINT]
rename_existing: ORIG_CANCEL_PRINT
gcode:
  <your custom gcode>
```

In this scenario, your custom macro is called when `CANCEL_PRINT` executes.  The built-in `CANCEL_PRINT` is renamed to `ORIG_CANCEL_PRINT`.

KevinOConnor, 2020-08-09:
FWIW, I think CANCEL_PRINT could be a bit confusing for the functionality of aborting a pause.  Maybe we should rename the command to PAUSE_CANCEL?

-Kevin

AbomShep, 2020-08-09:
Thanks Arksine, I'll go with that. I think Keven has a good point too.

Arksine, 2020-08-10:
@KevinOConnor The CANCEL_PRINT gcode can be used to abort a print even if it isn't paused.  By default it executes a PAUSE, sends "action:cancel" to Octoprint if `pause_resume` detects the source as such, then clears the paused state.  It should probably call "SDCARD_RESET_FILE" if it detects SD as the source.  

That said, if you think we should rename it I am ok with that.  Perhaps this gcode doesn't belong in pause_resume.py at all?

KevinOConnor, 2020-08-10:
Oh, sorry, I did not realize CANCEL_PRINT was a generic way to cancel a print.  I'm okay with leaving it as it is, but it's your call.

-Kevin

AbomShep, 2020-08-10:
Looks like my reply isn't here yet, but I didn't word it well. We have the option of performing custom gcode when the filament runs out, and when it's replaced. I wouldn't want klipper to make it's own decisions outside of that in a simple pause command. Also, you mentioned Octoprint, but it's not the only software being used with Klipper, so that may be something to keep in mind. 
Now if you want to create a cancel command, what is it going to do, and how is that going to be communicated to the sections that are already in place such as Pause_resume, Filament_switch_sensor, M600 macros. Will we be able to code a behaviour to go a long with it etc. etc.

Arksine, 2020-08-10:
@KevinOConnor I think it makes sense to leave it as it provides a default for "cancel" operations.

@AbomShep The reason why "CANCEL_PRINT" is in the pause_resume module is because that module can detect the source of the print (over the pty, ie octoprint, vs the virtual_sdcard).  As mentioned above, including it provides default behavior for CANCEL_PRINT.   Those who wish to override it for custom behavior can do so.  The same is true of PAUSE and RESUME.  This makes more sense than adding options to every module that wish to override its behavior as it allows the user to configure it in one place rather than many.

closed 2020-09-04 by AbomShep