Klipper3d/klipper #3168: Delta Moves/Crashes into Bed after save_config following bed_mesh_calibrate

lingo124, 2020-08-09:
After running bed_mesh_calibrate, I run save_config. Right before I send that command, the hot end is just sitting maybe 5 cm above the bed. Once that command is sent, the whole hot end moves down into the bed. I am assuming it only stops because the probe is triggered. Not sure why this occurs. This behavior does not occur after probe_calibrate.

Note, I am using Klipper on a raspberry pi 4 with a delta printer.
(https://github.com/KevinOConnor/klipper/files/5048245/klippy.log)

Arksine, 2020-08-10:
This is definitely abnormal behavior, nothing in BED_MESH_CALIBRATE nor SAVE_CONFIG should move the tool.  I would recommend pulling the latest code and then attempt to reproduce.  If after SAVE_CONFIG the tool starts moving issue an emergency stop gcode (M112), then upload the resulting log.  This would provide an info dump that would give us a better idea of what is going on.

mcmatrix, 2020-08-19:
I see that you have delta printer?

Could it be that after `SAVE_CONFIG` the klipper is restarted so that config can be re-read.
During restart tower motors are turned off and the printhead will drop to bed by it's own weight.

lingo124, 2020-10-02:
Sorry been a while. Did not have time to do troubleshooting. I just installed the latest klipper. The crashing still happens. I have tried to get the log immediately after issuing the save config command using the M112 command. Please see attached.

Also note, with the power off, I can pick up the effector up off the build surface maybe 5 mm and it will stay there, so I don't think the weight is causing it to fall down.

This is what is read out in terminal at the end of this process.

`Recv: // probe at -0.005,99.992 is z=15.462733
Recv: // Mesh Bed Leveling Complete
Recv: // Bed Mesh state has been saved to profile [default]
Recv: // for the current session.  The SAVE_CONFIG command will
Recv: // update the printer config file and restart the printer.
Recv: ok
[...]
Send: SAVE_CONFIG
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Closing serial port due to emergency stop M112.
Recv: // Klipper state: Shutdown
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline (Error: Closing serial port due to emergency stop M112.)"`

[klippy.log](https://github.com/KevinOConnor/klipper/files/5319183/klippy.log)

Arksine, 2020-10-08:
I don't see any indication in the log that the steppers were commanded to move after `SAVE_CONFIG`.   I do see where I believe the drivers are disabled.

Have you tried commanding the tool to X0 Y100 Z30 and issuing an M84?  If the head drops then it is certain that its a gravity issue.  If it doesn't drop then it would seem that somehow the steppers are driving the tool down.

lingo124, 2020-10-16:
So with other testing, I did manage to cause the head to drop with an M84 with the head in that position. It is rather odd because previously I had tested with the power off and the head can hold that position. With the machine powered on and in that position, when I disabled to stepper (which I assume equates to a power off), I would have expected the behavior to be similar. Maybe I'm missing something about how disabling the stepper differs from having it powered off, but at least in this case I think the issue is known and is just a stepper disable issue. I guess for probe calibrate or delta calibrate, a M84 isn't issued because this does not happen (also since probe calibrate is in the center too so... yah).

closed 2020-11-14 by github-actions[bot]