Klipper3d/klipper #3170: Kilpper will not reconnect after FIRMWARE_RESTART

dparker2112, 2020-08-09:
Hi all, I have a cranky BIQU Kossel!  My adventure has stalled today 🙁  Hopefully someone here can help.
It now has a Bigtree SKR 1.4 turbo and am running Octopi 1.4.2 on a pi 3B+, also just updated to python 3.  Everything works, meaning the board boots up. LCD readout ok, can connect from Octoprint, home, move etc.  Am editing the .cfg and when I FIRMWARE_RESTART it won't reconnect unless I reboot the pi

Recv: ok B:20.1 /0.0 T0:21.0 /0.0
Send: firmware_restart
Recv: ok
Send: M105
Recv: ok B:20.1 /0.0 T0:21.0 /0.0
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0Send: G91
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: G28 Z0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: G90
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5048574/klippy.zip)

KevinOConnor, 2020-08-10:
See #2156 .  As far as we know, Klipper works as intended, but some of these boards have flaky bootloaders.

-Kevin

dparker2112, 2020-08-10:
Yes, can confirm that using only RESTART works - thanks.  I'm kind of new at this, am I reading correctly in #2156 that potentially, reflashing the bootloader can clear this problem up, at the risk of bricking the board?  It seems once I get the .cfg done I shouldn't need to restart the firmware that much?

KevinOConnor, 2020-08-18:
Alas, I don't have that board and thus I don't personally know the answer to your questions.  It's generally rare to require a firmware restart, but it can be quite inconvenient if the automatic restart isn't working.

-Kevin

dparker2112, 2020-08-18:
Thanks, all up and running now.  Ditched the Octoklipper plugin and restart the firmware with the big button on the power supply.

No desire at all to mess with the bootloader at this point!

⁣About David Parker

Sent from BlueMail ​

closed 2020-09-09 by dparker2112