Klipper3d/klipper #3172: Unknown command:"M108"

pangklee, 2020-08-10:
Just installed klipper and ran a few calibration cubes just fine. Tried to calibrate pressure advance and print the square tower. Now all my prints are getting this error even the existing calibration cubes that printed successfully. 

start gcode
```
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder

M140 S{material_bed_temperature_layer_0} ;Start heating bed
M190 S{material_bed_temperature_layer_0} ;Wait for bed to reach temp before proceeding

M104 S{material_print_temperature_layer_0} ;Set Hotend temperature

G28 ; Home all axes
M109 S{material_print_temperature_layer_0} ;Wait for Hotend to reach temp before proceeding

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X5.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X5.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X5.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

looked through the log file and saw the below. Not sure why this would cause an issue.
```
Move out of range: 5.100 20.000 -0.035 [0.000]
Unknown command:"M108"
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/5053435/klippy.log)

BlackStump, 2020-08-10:
[If using "probe:z_virtual_endstop" then do not define a position_endstop in the stepper_z config section.](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L591)
[instead use position_min: -0.5](https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L41)
-0.5 is a example depending on how warped your bed is, bed mesh and probing may need to dip below 0.0 to cater for bed warp.
 >Unknown command:"M108"

Is most likely coming from Octoprint and will not cause any issues but you may want to configure Octoprint to be more Klipper friendly.

pangklee, 2020-08-10:
Looks like I had a duplicate entry with one commented out... I removed the position_endstop and everything works. Closing the issue.
Thanks

closed 2020-08-10 by pangklee