Klipper3d/klipper #3173: [Question] Scheduling no movement on a stepper

RadicalZephyr, 2020-08-11:
Hi Kevin,

I'm playing around with using the Klipper firmware to manipulate steppers using my own control program, and I'm trying to understand how to communicate to the firmware that I want one stepper to move but not another.

According to the developer docs, each step time interval is a relative time offset to the time the previous step was executed. It seems to me that one of two methods might be used to indicate that a given stepper had no motion. One method I can see would be to send a `queue_steps` command with a count of 0 to the non-moving stepper with the same interval as the total duration of all the steps of the moving stepper. The other method be that the next step to be scheduled on the paused stepper would use as it's first interval the entire time for that stepper to pause to let the movement of the moving stepper complete.

I've been looking through the step-generation code in the python and chelper trying to figure out where exactly this calculation is taking place and I'm having a hard time following it. If you could let me know what gets generated when a stepper has no movement to keep it in sync with the timings of the other steppers that would be really helpful.

Thanks!

KevinOConnor, 2020-08-11:
> The other method be that the next step to be scheduled on the paused stepper would use as it's first interval the entire time for that stepper to pause to let the movement of the moving stepper complete.

That.

Be aware that generating the `queue_step` commands manually is a bit of work.  The Klipper host software generates a list of step times and then uses an automated algorithm for generating the appropriate commands from those times (`klippy/chelper/stepcompress.c:compress_bisect_add()`).

Cheers,
-Kevin

RadicalZephyr, 2020-08-11:
Ah, awesome thank you!  Thanks for the pointer to where that compression is happening I'll take a look at that too.

closed 2020-08-11 by RadicalZephyr

KevinOConnor, 2020-08-11:
FYI, be sure to also see the docs at: https://www.klipper3d.org/Code_Overview.html#code-flow-of-a-move-command

-Kevin