Klipper3d/klipper #3174: Strange "Move out of range" issue

Sineos, 2020-08-11:
My printer was working quite well but then I decided to improve my bed_mesh and after this the printer suddenly started to run through the print-start-macro and then fail with a `Move out of range` error.

I found a work-around and boiled this down to few lines of gcode to reproduce the issue:

**Fails:**
```
G28
G90
G1 X0.4 Y30 Z0.3 F1500.0; End of the priming move from the start macro
G1 Z0.200 F480; Here the slicer generated gcode starts
G1 X86.475 Y85.581 F6000
M117 ThisFails
```

**Works:**
```
G28
G90
G1 X0.4 Y30 Z0.3 F1500.0; End of the priming move from the start macro
G1 X122 Y93 Z1; Add a move away from the offending coordinates
G1 Z0.200 F480; Here the slicer generated gcode starts
G1 X86.475 Y85.581 F6000
M117 ThisWorks
```

As you can see, this is both essentially the same gcode, with the only difference that I added an additional move away from the problematic position in line 4 of the working version.

The attached klippy.log contains both gcode snippets:
- Working: Line 327 - 371
- Failing: Line 381 - 413

[klippy.log](https://github.com/KevinOConnor/klipper/files/5055037/klippy.log)

schnello, 2020-08-11:
What i miss in your config is a "position_min: -0.5" in your [stepper_z] section.

Sineos, 2020-08-11:
Thanks, this did it. RTFM would have helped

closed 2020-08-11 by Sineos