Klipper3d/klipper #3175: MCU 'mcu' shutdown: Rescheduled timer in the past

Johann772, 2020-08-11:
I'm getting "MCU 'mcu' shutdown: Rescheduled timer in the past" after about two hours into an 8 hour print. My controller board is an Arduino Mega with Ramps 1.4 shield. I'm printing at 80 mm/s. I'm using PrusaSlicer as slicing software. I would greatly appreciate if someone can look at my Klipper log and also G-code file attached and tell me what could cause this issue. I've been using Klipper successfully for about three years (printing at 40 mm/s). I recently decided to increase the speed to 80 mm/s and successfully printed smaller parts (1 hour prints). 

If this issue is caused by the slicer software, is it possible to scan and check the G-code file via some software and eliminate this issue before starting the print?

[klippy2.zip](https://github.com/KevinOConnor/klipper/files/5057874/klippy2.zip)
[gcode.zip](https://github.com/KevinOConnor/klipper/files/5057956/gcode.zip)

KevinOConnor, 2020-08-11:
A "Rescheduled timer in the past" error indicates the micro-controller was unable to keep up with the requested schedule of events.

Was this a one-time error, or has this been happening repeatedly?  If it reoccurs, does it reliably occur in some particular set of prints?

Unfortunately, there's no way to check for this type of error in the host software, as the error occurred internally in the micro-controller.

Typically, the problem occurs when one is requesting too many steps per second of a given micro-controller.  However, your setup and load measurements aren't generally indicative of this type of problem.  Here's a graph of the load from your log (see docs/Debugging.md if curious):

[image]

It's also possible the micro-controller is having an internal fault of some kind (eg, overheating, noise on its crystal, emi) that is manifesting in the above error.  That's a long shot though.

-Kevin

Johann772, 2020-08-11:
Thanks for the quick response Kevin. This was not a one time issue, it actually happened six times in a row, but not consistently at any particular time into the print. I also tried to reproduce the error by using the "dry print" plugin for Octoprint (with heater and extruder commands bypassed), but interestingly was unable to reproduce it. I don't know if perhaps the heater and/or extruder commands could cause this issue somehow?

I agree with you that it could be a problem with the micro controller. I will replace the micro controller and report back.

KevinOConnor, 2020-08-13:
Is it always a "Rescheduled timer in the past" error, or do you sometimes get different errors?

-Kevin

Johann772, 2020-08-13:
No, I also received "Move queue empty".

KevinOConnor, 2020-08-14:
Ah, okay.  A "move queue empty" error would generally have a different root-cause then "rescheduled timer in the past".  FWIW, getting different errors like that tends to indicate general micro-controller instability as the likely culprit.

-Kevin

Johann772, 2020-08-14:
Thanks Kevin, I've ordered a new Micro controller and I believe that should fix the problem.

Johann772, 2020-08-15:
The new micro controller fixed the problem. I was able to print the same G-code file with no problems. @KevinOConnor thanks for your inputs and help to resolve this issue.

closed 2020-08-15 by Johann772

n8bot, 2020-08-15:
Do you mean that you replaced the Arduino/Ramps with another identical unit, or a different model altogether?