Klipper3d/klipper #3176: "MCU 'mcu' shutdown: Timer too close" when trying to access menu

kevzor, 2020-08-12:
Ever since the latest update I've been getting a "MCU 'mcu' shutdown: Timer too close" error randomly when trying to open the menu by pressing the scroll wheel button on my Ender 3 with the Creality 1.1.5 silent board. 
I've got Klipper running on a Pi 3B+ and I've tried reflashing the firmware but that didn't fix the issue. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5060108/klippy.log)

KevinOConnor, 2020-08-13:
@mcmatrix - any thoughts on this?

-Kevin

mcmatrix, 2020-08-14:
@KevinOConnor - I can check but I'm not sure I know how to debug such issue.
What I have noticed is that sometimes I'll get a `Failed to home a: Timeout during homing` 
error when I use menu and buttons during long homing.

Are these to issues related, at the moment I don't know?
Maybe you have some ideas about what symptoms these might be.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5073435/klippy.log)

KevinOConnor, 2020-08-15:
The likely reason for both "timer too close" and "no next step" (your "timeout during homing") errors is high host cpu usage resulting in the host being unable to send timely messages to the micro-controller.  Is there something in the menu code that is doing significant processing and/or could be looping for a long time?

-Kevin

mcmatrix, 2020-08-17:
@KevinOConnor 

> Is there something in the menu code that is doing significant processing and/or could be looping for a long time?

Every time when the menu is opened on the screen, it'll populate and update the items (menu manager `begin` method).
I did some measurements. 
What is a long time for Klipper host?

During homing, I toggled the main menu open and close.
```
menu:begin: elapsed: 216.95 ms
menu:begin: elapsed: 256.79 ms
menu:begin: elapsed: 177.51 ms
menu:begin: elapsed: 253.43 ms
menu:begin: elapsed: 171.69 ms
menu:begin: elapsed: 235.19 ms
Stats 1834230.4: gcodein=968 mcu: mcu_awake=0.002 mcu_task_avg=0.000007 mcu_task_stddev=0.000007 bytes_write=30617 bytes_read=19960 bytes_retransmit=9 bytes_invalid=0 send_seq=742 receive_seq=742 retransmit_seq=5 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99997259 heater_bed: target=0 temp=27.0 pwm=0.000 print_time=291.037 buffer_time=0.000 print_stall=0 extruder: target=0 temp=27.1 pwm=0.000 sysload=0.33 cputime=6.731 memavail=610104
menu:begin: elapsed: 225.69 ms
MCU 'mcu' shutdown: No next step
```

I was using following approach for time debugging.
```py
class MenuTimer(object):
    '''Measure the elapsed time in a code block.'''
    def __init__(self, printer, prefix, verbose=True):
        self.reactor = printer.get_reactor()
        self.prefix = prefix
        self.verbose = verbose

    def __enter__(self):
        self.start = self.reactor.monotonic()
        return self

    def __exit__(self, type, value, traceback):
        self.end = self.reactor.monotonic()
        self.elapsed = self.end - self.start
        if self.verbose:
            logging.info(
                '%s: elapsed: %.2f ms', self.prefix, (self.elapsed * 1000))

```

MenuManager methods
```py
    def catchtime(self, prefix, verbose=True):
        return MenuTimer(self.printer, prefix, verbose)
```
```py
    def _click_callback(self, eventtime, event):
        if self.is_running():
            self.press(event)
        else:
            # lets start and populate the menu items
            with self.catchtime('menu:begin'):
                self.begin(eventtime)
```

Also measured menu `rendering` and `press`
```
Starting heater checks for heater_bed
Starting heater checks for extruder
menu:begin: elapsed: 259.96 ms
menu:render&draw_text: elapsed: 1.65 ms
menu:render&draw_text: elapsed: 1.33 ms
menu:render&draw_text: elapsed: 1.33 ms
menu:render&draw_text: elapsed: 1.34 ms
menu:render&draw_text: elapsed: 3.45 ms
menu:render&draw_text: elapsed: 3.13 ms
menu:render&draw_text: elapsed: 2.96 ms
[... 165 lines cut ...]
menu:render&draw_text: elapsed: 1.88 ms
menu:render&draw_text: elapsed: 1.77 ms
menu:render&draw_text: elapsed: 1.75 ms
menu:render&draw_text: elapsed: 1.92 ms
menu:render&draw_text: elapsed: 1.90 ms
menu:render&draw_text: elapsed: 2.09 ms
menu:render&draw_text: elapsed: 2.02 ms
menu:press: elapsed: 4.69 ms
menu:render&draw_text: elapsed: 3.85 ms
menu:render&draw_text: elapsed: 1.42 ms
menu:render&draw_text: elapsed: 1.41 ms
menu:render&draw_text: elapsed: 1.69 ms
menu:render&draw_text: elapsed: 1.42 ms
menu:render&draw_text: elapsed: 1.95 ms
menu:render&draw_text: elapsed: 1.43 ms
menu:render&draw_text: elapsed: 1.40 ms
menu:render&draw_text: elapsed: 3.15 ms
menu:render&draw_text: elapsed: 3.11 ms
menu:render&draw_text: elapsed: 3.27 ms
menu:render&draw_text: elapsed: 3.18 ms
menu:render&draw_text: elapsed: 3.14 ms
menu:render&draw_text: elapsed: 3.75 ms
menu:render&draw_text: elapsed: 3.09 ms
menu:press: elapsed: 1.10 ms
```

KevinOConnor, 2020-08-17:
> What is a "long time" for Klipper host?

I'd say anything more than about 10ms.

The "render&draw_text" doesn't appear to be a problem.  We only need to draw the screen every 500ms or so.  We can add a `display.request_redraw()` to request faster updates on a key press.

Do you know why "menu:begin" is taking so much time?

-Kevin

mcmatrix, 2020-08-17:
> Do you know why "menu:begin" is taking so much time?

I'll investigate that part of code and try to optimize it.

-Janar

KevinOConnor, 2020-08-17:
FWIW, I think it would be helpful to simplify `MenuManager` by adding `MenuList.draw()` and `MenuList.key_event()` methods, and then have MenuManager just invoke those functions with the currently active menu.  (That is, I think it may make the code simpler if each menu "knew how to draw itself".)  I don't know if this would improve performance, but I think it would make the code simpler.

-Kevin

mcmatrix, 2020-08-18:
I have located the bottleneck, it's the menu container populate method. 
It requires to refactoring the related code. I'll post new PR when I'm done.

mcmatrix, 2020-08-18:
Hi @kevzor

Could you test this PR #3206
Are you able to get rid of your error?

mcmatrix, 2020-08-18:
@KevinOConnor 

I was moving inside the menu and after optimizations #3206 the measurements are:
```
Starting heater checks for heater_bed
Starting heater checks for extruder
'__main':populate took 1.05 ms
'__main __octoprint':populate took 0.65 ms
'__main __control':populate took 1.19 ms
'__main __control __move_01mm':populate took 0.61 ms
'__main __setup':populate took 0.86 ms
'__main __setup':populate took 1.68 ms
'__main __setup __tuning':populate took 0.45 ms
'__main __temp':populate took 0.86 ms
'__main __filament':populate took 0.72 ms
'__main':populate took 0.99 ms
'__main __octoprint':populate took 0.49 ms
```
Only the container `populate` method was measured.
When toggling menu on and off, I wasn't able to get the timeout error anymore.

mcmatrix, 2020-08-19:
Hi @kevzor

The #3206 was merged to master.
Please update your Klipper code and give it a try.

kevzor, 2020-08-20:
Thanks for looking into this so quickly. I updated Klipper this morning and just had another "Timer too close" error while going through the menus. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5101494/klippy.log)

mcmatrix, 2020-08-20:
@kevzor Thank you for the feedback.
Was printer printing from virt.sdcard?
Can you tell me what menu you were opening/entering when "Timer too close" happend?
Or what actions you did in menu?

kevzor, 2020-08-20:
I was paused in the middle of a virtual SD card print for a filament swap. I was already in the main menu and was entering the Control menu to insert the second filament after having already retracted the first filament from Octoprint, but when I clicked on the Control menu option I got the "Timer too close" error.

kevzor, 2020-08-20:
It seems to be consistently happening when I am paused mid-print for a filament swap and try to enter the SD Card menu to resume the print. I've noticed it takes a little bit longer to load the SD Card menu since the update as well. I'll attach my latest log from just after one of the SD Card menu errors.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5105719/klippy.log)

mcmatrix, 2020-08-21:
Thanks for the info.
How many gcode files you have in the virtual SD card folder?
I'm trying to do some performance tests and optimized if necessary.

mcmatrix, 2020-08-21:
@KevinOConnor I'll try to find more ways to improve existing code.
I already saw a way to improve the rendering part.

I was just wondering is it possible to somehow separate method that runs longer from
the main thread so that it won't block critical timings (timer errors) by using `reactor.register_async_callback`?

-Janar

closed 2020-09-19 by github-actions[bot]