Klipper3d/klipper #3179: BLTOUCH after bowden crash into bed !

xsasx, 2020-08-12:
_**After i chaned my anet a8 to bowden (with same bltouch holder) it crash into bed when I home Z

Must I do anything in the config  ? 

Here is my klipper config (dont blame me its a old but working version i run im scared to upgrade)**_

Please help :( it worked so great till bowden

```
# This file contains common pin mappings for Anet A8 printer from 2016
# and 2017. To use this config, the firmware should be compiled for
# the AVR atmega1284p.

# Note that the "make flash" command does not work with Anet boards -
# the boards are typically flashed with this command:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
step_distance: .01
endstop_pin: ^!PC2
position_endstop: -30
position_max: 220
position_min: -30
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: PC7
enable_pin: !PD6
step_distance: .01
endstop_pin: ^!PC3
position_endstop: -8
position_min: -8
position_max: 220
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PA5
step_distance: .0025
#endstop_pin: ^!PC4
endstop_pin: probe:z_virtual_endstop
#position_endstop: .5
position_min: -8
position_max: 240
homing_speed: 20

[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .010185
pressure_advance: 0.190
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA7

#control = pid
#pid_kp = 23.867
#pid_ki = 0.240
#pid_kd = 594.280
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA6
#control = pid
#pid_kp = 76.824
#pid_ki = 1.330
#pid_kd = 1109.152

min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
baud: 115200
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 1000
max_z_velocity: 20
max_z_accel: 100

[display]
lcd_type: hd44780
rs_pin: PA3
e_pin: PA2
d4_pin: PD2
d5_pin: PD3
d6_pin: PC0
d7_pin: PC1

[bltouch]
sensor_pin: PC4
control_pin: PA4
pin_move_time:1.0
#hoch ist tief und tief ist hoch (durckkopf nach oben bringen)
z_offset: 0
speed:8
x_offset: 60
y_offset: 0

#[servo bltouch]
# set pin to the pin that the bltouch
#servo is connected to.
#pin: PA4
#maximum_servo_angle: 180
#minimum_pulse_width: 0.0006
#maximum_pulse_width: 0.0024

# Define a probe using the BLTouch
#[probe]
#pin: ^PC4
#speed: 3.0
#x_offset: 60
#y_offset: 0.0
#z_offset: 1.821
#activate_gcode:
   # SET_SERVO SERVO=bltouch ANGLE=10
   # SET_SERVO SERVO=bltouch ANGLE=60
  #  G4 P200
#deactivate_gcode:
  #  SET_SERVO SERVO=bltouch ANGLE=90

[homing_override]
axes: z
gcode:
    G90
	G0 Z20
    G1 Z7 F600
    G28 X0 Y0
    G1 X110 Y125 F3600
    G28 Z0
set_position_z: 5

[bed_mesh]
speed: 50
horizontal_move_z: 5
samples: 3
sample_retract_dist: 2.0
min_point: -30,10
max_point: 150,210
probe_count: 8,8
fade_start: 1.0
fade_end: 10.0
split_delta_z: 0.025
move_check_distance: 5.0
mesh_pps: 3,3
algorithm: bicubic
bicubic_tension: .2

# Heater and temperature sensor verification. Heater verification is
# automatically enabled for each heater that is configured on the
# printer. Use verify_heater sections to change the default settings.
[verify_heater extruder]
heating_gain: 1
#   The minimum temperature (in Celsius) that the heater must increase
#   by when approaching a new target temperature. The default is 2.
check_gain_time: 90
#   The amount of time (in seconds) that the heating_gain must be met
#   in before an error is raised. The default is 20 seconds for
#   extruders and 60 seconds for heater_bed.
hysteresis: 20
#   The difference between the target temperature and the current
#   temperature for the heater to be considered within range of the
#   target temperature. The default is 5.
max_error: 150
#   The maximum temperature difference a heater that falls outside the
#   target temperature range may accumulate before an error is
#   raised. For example, if the target temperature is 200, the
#   hysteresis is 5, the max_error is 120, and the temperature is
#   reported at 185 degrees for 12 seconds then an error would be
#   raised (or 24 seconds at 190, or 120 seconds at 194, etc.). The
#   default is 120.

# Heater and temperature sensor verification. Heater verification is
# automatically enabled for each heater that is configured on the
# printer. Use verify_heater sections to change the default settings.
[verify_heater heater_bed]
heating_gain: 1
#   The minimum temperature (in Celsius) that the heater must increase
#   by when approaching a new target temperature. The default is 2.
check_gain_time: 30
#   The amount of time (in seconds) that the heating_gain must be met
#   in before an error is raised. The default is 20 seconds for
#   extruders and 60 seconds for heater_bed.
hysteresis: 5
#   The difference between the target temperature and the current
#   temperature for the heater to be considered within range of the
#   target temperature. The default is 5.
max_error: 150
#   The maximum temperature difference a heater that falls outside the
#   target temperature range may accumulate before an error is
#   raised. For example, if the target temperature is 200, the
#   hysteresis is 5, the max_error is 120, and the temperature is
#   reported at 185 degrees for 12 seconds then an error would be
#   raised (or 24 seconds at 190, or 120 seconds at 194, etc.). The
#   default is 120.

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 26.056
#*# pid_ki = 1.135
#*# pid_kd = 149.499
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 74.611
#*# pid_ki = 1.330
#*# pid_kd = 1046.419
#*#
#*# [bed_mesh default]
#*# points =
#*# 	-0.325000, -0.195000, -0.025000, 0.090833, 0.101667, 0.053333, 0.060833, 0.187500
#*# 	-0.334167, -0.180000, -0.025000, 0.110833, 0.146667, 0.064167, 0.065833, 0.238333
#*# 	-0.290000, -0.170000, -0.018333, 0.109167, 0.124167, 0.055833, 0.063333, 0.205000
#*# 	-0.228333, -0.158333, -0.021667, 0.116667, 0.080833, 0.040833, 0.087500, 0.170833
#*# 	-0.259167, -0.145000, -0.012500, 0.117500, 0.104167, 0.005833, -0.006667, 0.159167
#*# 	-0.249167, -0.153333, -0.010000, 0.120833, 0.085833, 0.015000, 0.002500, 0.181667
#*# 	-0.277500, -0.110833, -0.015000, 0.098333, 0.096667, 0.025833, -0.015000, 0.144167
#*# 	-0.289167, -0.127500, -0.024167, 0.081667, 0.061667, 0.010000, -0.000000, 0.025833
#*# x_count = 8
#*# y_count = 8
#*# min_x = -30.0
#*# max_x = 149.97
#*# min_y = 10.0
#*# max_y = 209.99
#*# x_offset = 60.0
#*# y_offset = 0.0
#*# mesh_x_pps = 3
#*# mesh_y_pps = 3
#*# algo = bicubic
#*# tension = 0.2

```

BeatSlayer, 2020-08-12:
Recheck your wiring.

xsasx, 2020-08-13:
> Recheck your wiring.

dont thinnk it has anything todo with wirering ? pin up and down work - also the setup worked now for years - i just use now bowsden instead of direct. but it wont work and crash into bed ? maybe i must do any setting cause of different position of the nozzle or new messure of Z ? or something like this ?

BlackStump, 2020-08-13:
[I suggest you verify your BLTouch is functioning](https://www.klipper3d.org/BLTouch.html)

thomask77, 2020-08-14:
I would also suggest to Test your BLtouch.
If PROBE_QUERY always reports "open", you may be missing a Pull-Up on you sensor_pin.

Try `sensor_pin: ^PC4`

closed 2020-09-13 by github-actions[bot]