Klipper3d/klipper #3180: Ender 3 Pro with the STM 4.2.2 Board?

r1rider01, 2020-08-12:
Has anyone gotten this to work yet?   

Not sure the 'make menuconfig' settings to set, but i've tried STM32F103, with 28kb bootloader, and still unable to get it to flash either from SSH or by copying the .bin file to the SD and loading that way.

When I do copy the .bin to SD and place in the 3d printer, the screen goes blank and just get the blue backlight, but klipper/octoprint are still unable to talk to it.

I know this board is pretty new on the E3Pro so wasn't sure if anyone has seen it yet?

Thanks.

manu7irl, 2020-08-13:
This is the same mcu as the ender-3 v2.
There is already a config file for it...
Take a look, there.

r1rider01, 2020-08-13:
Where can I find the bin file?  Or what settings are needed to make the bin in menu config?

r1rider01, 2020-08-13:
Ok I see what you mean by the config.  I followed these steps:

# This file contains pin mappings for the stock 2020 Creality Ender 3
# V2. To use this config, during "make menuconfig" select the
# STM32F103 with a "28KiB bootloader" and with "Use USB for
# communication" disabled.

# Flash this firmware by copying "out/klipper.bin" to a SD card and
# turning on the printer with the card inserted. The firmware
# filename must end in ".bin" and must not match the last filename
# that was flashed.

Still no go on flashing....but Marlin2 flashes right away from the SD card

closed 2020-08-20 by klipper-gitissuebot

Ducati87, 2021-02-12:
same problem. Any fix? Thank you.

CoolPickle, 2021-11-17:
Same problem, please fix this!