Klipper3d/klipper #3182: Dyze 500 thermistor

kakou-fr, 2020-08-13:
I am trying to use a Dyze 500 (from Mellow) thermistor.

According to [https://dyzedesign.com/support/temperature-sensors/](https://dyzedesign.com/support/temperature-sensors/) i defined : 

```
[thermistor dyze500]
temperature1: 25.
resistance1: 4500000
temperature2: 200.
resistance2: 8070
temperature3: 400.
resistance3: 253
```

but my temperature is always  negative

[klippy.log](https://github.com/KevinOConnor/klipper/files/5069456/klippy.log)

KevinOConnor, 2020-08-13:
It looks like you've configured it correctly.  What does the resistance show if you hook the thermistor up to a multimeter?  What does the `QUERY_ADC` command report (temporarily use `min_temp=-200` if needed).

-Kevin

randellhodges, 2020-08-17:
Same issue with the TL clone:
https://www.aliexpress.com/item/4000907614466.html

`Recv: ok FIRMWARE_VERSION:v0.8.0-715-g9d9e5c7e FIRMWARE_NAME:Klipper`

```
[thermistor Trianglelab_T-D500]
## https://www.aliexpress.com/item/4000907614466.html
temperature1: 25
resistance1: 4721000
temperature2: 200
resistance2: 8000
temperature3: 400
resistance3: 264.6
# R(200C)     = 8K +- 5%
# B(150/250)  = 5300k +- 3%
```

Room temp, around 26C and the temp is bouncing around -20C and -10C.

Meter reads 2.6M Ohm at 26C (seems like these thermistors aren't reading what their tables says they should read?)

```
Send: QUERY_ADC NAME=extruder
Recv: // ADC object "extruder" has value 0.999817 (timestamp 1622.770)
```

Connected to an SKR 1.4 board

KevinOConnor, 2020-08-17:
FYI, these sensors will have very poor accuracy at temperatures below ~75C.  The hardware ADC doesn't have enough precision to measure the minuscule voltage changes the sensor will create at these low temperatures.

For reference, here's a chart of ADC resolution for a number of common temperature sensors:
[image]

Note how low the "ADC change per 1C" is for the "Dyze 500" sensor for temperatures below ~75C.

That said, the sensor may still work okay for an extruder (where working temperatures are typically above 100C).

You may want to build temperature/resistance pairs for your specific sensor.  (Heat the unit to a stable temperature, use the QUERY_ADC command to measure the resistance, and use an external temperature probe to measure the actual temperature.  Repeat for three different temperatures, with all three above 75C.)

-Kevin

randellhodges, 2020-08-17:
I appreciate the response.

I have a slice engineering thermistor running on an SKR 1.4 with Marlin and it has a reasonable room temp, only off by a couple of degrees. Maybe an unfair comparison because they have different values at room temp.

Oh, I see from your image, the change on the Dyze under 50C basically is flat while the Slice actually has some movement. That would probably explain the difference.

General question, for the LPC1768, are you using all 12 bits for the ADC? I'm assuming yes.

KevinOConnor, 2020-08-17:
> General question, for the LPC1768, are you using all 12 bits for the ADC?

Yes, along with 8x oversampling.  The ADC sensors just aren't capable of measuring that kind of minuscule voltage change though.  Also, the ADC sensors are generally less accurate at their extremes (near 0.0 and 1.0).

-Kevin

closed 2020-09-19 by github-actions[bot]

ZhuJo, 2021-01-30:
Can this issue be reopened again? I have same Trianglelabs thermistor T-D500 and it worked very nice in Marlin, never went below 21 °C and provided no errors. In Klipper I have to have min_temp: -10 to prevent Klipper against auto shutdown, which I don't consider safe. Can we have more than 3 values in custom temp table so we can provide more data?

lunatic1972, 2021-02-17:
I have not yet completed my build of Voron 2.4 and I am also using the same T-D500 Dyze sensors. Has this issue been addressed with adding more values or a different way to use custom sensors?

ZhuJo, 2021-02-17:
To me, using Beta curve works:

[thermistor Trianglelab_T-D500]
temperature1: 200
resistance1: 8000
beta: 5300

or for 3 temp/resistance values set for temperature1: 25, resistance1: 10000000 (yes, 10 million ohms), then values for 200 °C and 400 °C

VasilNikolov, 2021-07-28:
> Can this issue be reopened again? I have same Trianglelabs thermistor T-D500 and it worked very nice in Marlin, never went below 21 °C and provided no errors. In Klipper I have to have min_temp: -10 [...]

You can use the feature adc_temperature to specify the whole table of values for the thermistor.

VasilNikolov, 2021-07-28:
Here is a table I generated from the trianglelab documentation

`[adc_temperature Trianglelab_T-D500]
temperature1: 25
resistance1: 4721000.00
temperature2: 100
resistance2: 147700.00
temperature3: 110
resistance3: 104200.00
temperature4: 120
resistance4: 74560.00
temperature5: 130
resistance5: 54130.00
temperature6: 140
resistance6: 39810.00
temperature7: 150
resistance7: 29650.00
temperature8: 160
resistance8: 22340.00
temperature9: 170
resistance9: 17020.00
temperature10: 180
resistance10: 13100.00
temperature11: 190
resistance11: 10190.00
temperature12: 200
resistance12: 8000.00
temperature13: 210
resistance13: 6337.00
temperature14: 220
resistance14: 5062.00
temperature15: 230
resistance15: 4077.00
temperature16: 240
resistance16: 3309.00
temperature17: 250
resistance17: 2705.00
temperature18: 260
resistance18: 2228.00
temperature19: 270
resistance19: 1846.00
temperature20: 280
resistance20: 1540.00
temperature21: 290
resistance21: 1293.00
temperature22: 300
resistance22: 1092.00
temperature23: 310
resistance23: 929.80
temperature24: 320
resistance24: 795.80
temperature25: 330
resistance25: 684.20
temperature26: 340
resistance26: 590.80
temperature27: 350
resistance27: 512.20
temperature28: 360
resistance28: 445.80
temperature29: 370
resistance29: 389.40
temperature30: 380
resistance30: 341.20
temperature31: 390
resistance31: 300.00
temperature32: 400
resistance32: 264.60
temperature33: 410
resistance33: 234.00
temperature34: 420
resistance34: 207.50
temperature35: 430
resistance35: 184.50
temperature36: 440
resistance36: 164.50
temperature37: 450
resistance37: 146.90
temperature38: 460
resistance38: 131.60
temperature39: 470
resistance39: 118.10
temperature40: 480
resistance40: 106.10
temperature41: 490
resistance41: 95.60
temperature42: 500
resistance42: 86.30`

dsloyer74, 2021-12-19:
adc_temperature does not work.  With precise temp/resistance from Dyze Engineering I get -1582.7C at 26C room temp.

[adc_temperature DyzEndPro_500C]           #based on https://docs.dyzedesign.com/general-support/temperature-sensors.html#_500%C2%B0c-thermistor
temperature1: 10
resistance1: 8100000
temperature2: 20
resistance2: 5200000
temperature3: 25
resistance3: 4500000
temperature4: 30
resistance4: 2830000
temperature5: 60
resistance5: 666000
temperature6: 80
resistance6: 288000
temperature7: 100
resistance7: 136000
temperature8: 120
resistance8: 70000
temperature9: 140
resistance9: 38500
temperature10: 160
resistance10: 22200
temperature11: 180
resistance11: 13200
temperature12: 200
resistance12: 8070
temperature13: 220
resistance13: 5120
temperature14: 240
resistance14: 3380
temperature15: 260
resistance15: 2240
temperature16: 280
resistance16: 1555
temperature17: 300
resistance17: 1100
temperature18: 320
resistance18: 790
temperature19: 340
resistance19: 578
temperature20: 360
resistance20: 434
temperature21: 380
resistance21: 333
temperature22: 400
resistance22: 253
temperature23: 420
resistance23: 196
temperature24: 440
resistance24: 157
temperature25: 460
resistance25: 125.4
temperature26: 480
resistance26: 102
temperature27: 500
resistance27: 83.7

dsloyer74, 2021-12-19:
The following works, but only with -200C min_temp

[thermistor DyzEndPro_500C]           #based on https://docs.dyzedesign.com/general-support/temperature-sensors.html#_500%C2%B0c-thermistor
temperature1: 25
resistance1: 4500000
temperature2: 260
resistance2: 2240
temperature3: 500
resistance3: 83.7

jremen, 2021-12-20:
It works for me fine with these (printed plenty of own designs from ABS and PLA):
[thermistor Trianglelab_T-D500]
# ### https://www.aliexpress.com/item/4000907614466.html
# ### [extruder] senzor_type Trianglelab_T-D500
temperature1: 200
resistance1: 8000
beta: 5300

dsloyer74, 2021-12-26:
Using this table from DyzEnd > 
https://docs.dyzedesign.com/general-support/temperature-sensors.html#_500%C2%B0c-thermistor

and this guide for Beta calc > 
https://www.ametherm.com/thermistor/ntc-thermistor-beta

I was able to find my precise beta value for the DyzEnd Pro 500C Thermistor

200C - 8070 ohm
440C - 157 ohm
Beta = 5538.95

jremen, 2022-01-05:
As my TriangleLabs D-500 thermistor is now out of order (not because of it, but because 3D printer failure and making huge plastic blob on whole hotend) I bought Ender 5 Plus and step by step, I tune it.
Anyway, in a process I bought Mellow copper heat block which has two holes, for glass and 3 mm thermistors/thermocouples.
With it, I was able to measure temp/resistance values (up to 295 °C) for my Mellow DyzeDesign 500 like thermistor with great accuracy.

So I took 3 value pairs (2 actually needed) and put it into Steinhart-hart calculator:
https://www.thinksrs.com/downloads/programs/therm%20calc/ntccalibrator/ntccalculator.html

With this, I was able to find my beta (5240) and resistance at 200 °C (6847.523 – and must say, those values are quite stable).

Now, my delta between measured temps from 100 °C up to 290 °C with Epcos 100K B57560G104F (Ender 5 Plus default thermistor) and that Dyze Design clone from Mellow is not larger than 2 °C, but more like 0.5 °C difference (if any) at temps higher than 180 °C. So at full scale, I'm pretty on spot.

Definitely suggest to do measurements this way and put T1, R1 and Beta values for this thermistor according to your measurements. Steinhart-hart model with 3 temp/resistance pairs for this thermistor WORKS fine at higher temps, but throws firmware restart at room temps.
With Beta it doesn't happen and Klipper don't go into shutdown mode.

ant0nyk1ng, 2022-02-04:
Using Marlin I am having similar issues but as said temps under 190deg C  are not needed. The main issue is normally it all works fine but if I put on the aircon I have to use the heat gun to warm up the hotend til it starts to read otherwise I get a temp fail on the first print of the day. 

Can I put false numbers in the thermistor table or remove numbers below say 30deg in the thermistor table to get around Marlin thinking it's a heating issue?

Gilabite, 2022-02-04:
@ant0nyk1ng if you're using marlin why are you asking this on klipper's github? Why dont you try asking on marlins?

ant0nyk1ng, 2022-02-04:
sorry yes I wrote in the wrong page delete

jremen, 2022-02-07:
Marlin has this type of thermistor predefined. Now, question is accuracy but I think that within 5 °C everybody is fine.