Klipper3d/klipper #3184: hall filament width sensor problem, flow back 100% after every change

alocin73, 2020-08-13:
I've having a problem with the hall filament width sensor.

It measure correctly the width but immediately after the flow is changed to 104%(for example) it return instantly to 100%.

A little extract of the measurament:
Filament width:1.714
Filament width:1.718
Filament width:1.713
Filament width:1.713
Filament width:1.718
Filament width:1.719
Filament width:1.714
Filament width:1.711
Filament width:1.706
Filament width:1.704
Filament width:1.716
Filament width:1.700
Filament width:1.711
Filament width:1.717
Filament width:1.678
Filament width:1.705
Filament width:1.712
Filament width:1.716
Filament width:1.707
Filament width:1.706

Less important, but can someone explain to me what changes need to be made to the menu after the change to jinja?

Thanks

[klippy.log](https://github.com/KevinOConnor/klipper/files/5071489/klippy.log)

mcmatrix, 2020-08-14:
Maybe @test3210-d can answer your first issue?

I can answer about the menu.
You don't need an `input` element, the value is read-only anyway.
Just add empty `command` item.
Like below and it will be added to the end of list.
```ini
[menu __main __filament __width]
type: command
name: Width: {'%.2f' % printer.hall_filament_width_sensor.Diameter}
```
If you want that to be the first item in the submenu then also add `index: 0` attribute

alocin73, 2020-08-15:
@mcmatrix Thanks, the menu now its working.

After some research i've seen that '&.2f' etc.. are a python way of formatting a number (in this case) or a string.
May be an explanation of wat they mean should be helpfull in the documentation.

I've tried also a klipper fresh install, but the flow always change back to 100% after every update, i hope it's only a visualization problem, i'm tryng to see if that reflect on the print.

test3210-d, 2020-08-15:
When sensor  is enabled, every time as extruder start new line extrude factor is updated from filament width to new extrude factor
If you change extruder factor from menu it immeditaly owerwrited by new extruder factor from sensor

alocin73, 2020-08-15:
Thanks, but i'm not talking of manual changes.
In the tune menu the percentage it's updated(automagically :-) ) when the filamenth width change, I'm quite sure when i installed the sensor, was updated and the value dosen't change until next measurament/filament width change.
Now it's updated and goes back to 100% immediately.

test3210-d, 2020-08-15:
Ok, You have first part of filament uncnown diameter, while filament go from sensor measure point to extruder and it diameter uncnown by default flow set as 100%  try add 
use_current_dia_while_delay = True 
to sensor config, in this case uncnown part of filament be update to current filament diameter

alocin73, 2020-08-15:
I've to admit that i was skeptical, but enabling it, the reading are as expected, it's also visible in the print(40x40 1 wall cube).
The layer look more consistent and the wall is 0,44/0,45mm as set in the slicer.
I thought ok I'm stupid..........

I did another test:
-Mesured the filament to be sure that the delay was gone.
-Disabled use_current_dia_while_delay.

The value stay 100% till the delay(109mm), then it start again changing the value back to 100% after every update.
I measured the wall thickness and this time is 0,39/0,40mm, the filament its less then 1,75mm

Some reading from the log:
Filament width:1.687
Filament width:1.682
Filament width:1.681
Filament width:1.683
Filament width:1.686
Filament width:1.689
Filament width:1.687
Filament width:1.686
Filament width:1.689
Filament width:1.690

alocin73, 2020-08-16:
I think I've found the problem.

In extrude_factor_update_event after the reading of the first item:

`if pending_position <= last_epos:`
`                    # Get first item in filament_array queue`
`                    item = self.filament_array.pop(0)`
`                    filament_width = item[1]`

It always assign filament_width = self.nominal_filament_dia, or = self.diameter if use_current_dia_while_delay is set:

`elif self.use_current_dia_while_delay:`
`                   filament_width = self.diameter`
`                else:`
`                    filament_width = self.nominal_filament_dia`

So with use_current_dia_while_delay true it seem to work but it read the actual measurament and not the one from the array.

I've made it work storing the value of the first item of the array and then apply self.diameter or self.nominal_filament_dia only if pending_position is not changed.

`if pending_position <= last_epos:`
`                    # Get first item in filament_array queue`
`                    item = self.filament_array.pop(0)`
`                    self.filament_width = item[1]`
`                else:`
`                   if self.use_current_dia_while_delay and self.firstEpos == pending_position:`
`                        self.filament_width = self.diameter`
`                    elif  self.firstEpos == pending_position:`
`                        self.filament_width = self.nominal_filament_dia `

I'm not a programmer, only a little experience with VBA, but i've modified mine hall_filament_width_sensor.py that way and it seem to works.

[hall_filament_width_sensor.py.txt](https://github.com/KevinOConnor/klipper/files/5080866/hall_filament_width_sensor.py.txt)

closed 2020-09-14 by github-actions[bot]